mavlink/qgroundcontrol #4065: Survey: Explode survey

AndKe, 2016-09-20:
The survey tool is great, however, in an sloping landscape, or over landscape with big elevation changes, one side may need to be at very different elevation than the other.
Could we have a "explode" feature that presented survey as a bunch of waypoints ? - that would allow to adjust altitude where needed.

DonLakeFlyer, 2016-09-20:
Then the next thing you'll ask for is a grid editing tool because you'll lose you mind tweaking all those altitudes :)

tubeme, 2016-09-22:
I would love grid editing tool and a 3D view for it :)

AndKe, 2016-09-23:
@DonLakeFlyer if I am asking much, then why not go all the way :)
get elevation data from http://viewfinderpanoramas.org/dem3.html or google, and automatically plan a mission relative to ground, inserting wp as needed over hills in survey area. 
Better than flying by relative altitude in autopilot, as we are never 100% sure when/if AP got all elevation data it needed befor BRLOS or Iridium flight.  :)

- but honestly, I just thought that was about temporary masking/ignoring the strat/end commands that define waypoints as a survey.

DonLakeFlyer, 2017-12-29:
I'm thinking of living without the whole "explode" feature being done by round tripping through the vehicle instead of adding new ui.

closed 2017-12-29 by DonLakeFlyer

AndKe, 2017-12-30:
...or support saving to the standard .txt file - which is not far from what a upload to vehicle does..

DonLakeFlyer, 2017-12-30:
> ...or support saving to the standard .txt file - which is not far from what a upload to vehicle does..

Yeah, I plan to get to that in 3.4