mavlink/qgroundcontrol #4067: Android add android.hardware.location.gps

dagar, 2016-09-20:
Action required: If your app requires GPS hardware to operate properly, you will need to explicitly add the "android.hardware.location.gps" uses-feature to your manifest.

What’s changing

We’re making a change on October 15th, 2016 that will affect apps targeting API version 21 (Android 5.0, Lollipop) or higher that use ACCESS_FINE_LOCATION but don't explicitly have the "android.hardware.location.gps" uses-feature. Going forward, these apps will be available to install on devices that don't have GPS hardware. In most cases this won't be an issue since Wi-Fi and Cell-ID based location provides high enough fidelity for the typical operation of these apps. However, any apps that require GPS hardware, such as GPS navigators, should explicitly add the "android.hardware.location.gps" uses-feature to their manifest.

If your app requires GPS to function properly and you do not include android.hardware.location.gps in your manifest declaration, your users may have a poor app experience.

Also, if you’re using the fused location provider and wish to receive the most accurate location samples from GPS (i.e. with PRIORITY_HIGH_ACCURACY), you must include the "android.hardware.location.gps" feature in your app’s manifest to ensure that Google Play only distributes your app to devices with GPS sensors.

You can read more about this change in the Android Developers Help Center.

Regards,

The Google Play Team

Up to 20 affected app(s) and version(s) are listed below. If you have more than 20 apps in your catalogue, make sure to check each app for this issue.

org.mavlink.qgroundcontrol   301000

DonLakeFlyer, 2016-09-29:
@dogmaphobic I don't really know how these manifest things work. But this would only seem to be needed to prevent an install if there is no gps on the device. Which is not what we want for QGC. At least that is what I think it is saying.

dagar, 2016-09-29:
With Qt the permissions are partially based on dependencies. Upon rereading this I don't think we need to do anything.

Here are the current permissions FYI

    android.permission.ACCESS_FINE_LOCATION
    android.permission.ACCESS_NETWORK_STATE
    android.permission.BLUETOOTH
    android.permission.BLUETOOTH_ADMIN
    android.permission.INTERNET
    android.permission.READ_EXTERNAL_STORAGE
    android.permission.WAKE_LOCK
    android.permission.WRITE_EXTERNAL_STORAGE

closed 2016-09-29 by dagar