mavlink/qgroundcontrol #4080: qgc v3.0.1 does not upgrade firmware on Pixracer

andrejpan, 2016-09-28:
My current firmware version on Pixracer is `1.4.1dev` and new stable release is `1.4.4`. After the upgrade procedure there is no difference.

Here is also a video: https://sendvid.com/xbrpi5bm#

DonLakeFlyer, 2016-09-28:
From looking at the video the correct firmware is being uploaded. Can you do this:
1) Boot QGC
2) Menu: Widgets / Mavlink Inspector
3) Connect to the board showing the wrong version
4) Tell me what the value is for AUTOPILOT_VERSION.flight_sw_version

andrejpan, 2016-09-28:
[image]

DonLakeFlyer, 2016-09-28:
I'll have to load up the 1.4.4 stable version and do some double-checking. From the inspector value the firmware isn't sending the right version. But QGC isn't quite showing the correct thing either.

DonLakeFlyer, 2016-09-29:
I can repro this. The firmware is reporting 1.4.1dev as the version so QGC is doing the right thing.

@LorenzMeier @dagar Can somebody take a look at this? Either the stable url is pointing to the wrong build, or 1.4.4 stable is reporting the wrong version.

dagar, 2016-09-29:
The firmware upload was broken and has been fixed in master. I'll look at getting updated stable builds.

dagar, 2016-09-29:
Actually the firmware upload was only broken in master/beta.

Was something in the version reporting changed in the last ~ 1.5 months?
https://s3.amazonaws.com/px4-travis/Firmware/stable/px4fmu-v4_default.px4

DonLakeFlyer, 2016-09-29:
QGC firmware upload was broken? When? Where's the fix?

dagar, 2016-09-29:
PX4 firmware upload to s3 was broken in master for about a month and I just fixed it. The break didn't make it to stable/release.
https://github.com/PX4/Firmware/commit/acc1cc3158d378d8774921a42964e910d6352f77

DonLakeFlyer, 2016-09-29:
Ah, get it now.

DonLakeFlyer, 2016-09-29:
Version code in QGC hasn't changed since it went it as far as I know. Closing this on the QGC side since QGC is doing the right thing. Either the firmware link is wrong, or the firmware itself it reporting the wrong version. This should move to the firmware issues if this is a real problem.

closed 2016-09-29 by DonLakeFlyer