mavlink/qgroundcontrol #4081: Mission planner interface not working for generic protocol

mattmccrink, 2016-09-28:
I pulled the latest commit (9/28/16) and it seems the mission planner interface does not work with the generic flight vehicle protocol. I tested in debug mode with mock vehicles and in release mode with our stack (custom). 

Waypoint icons are not displayed, and editing can only be carried out in the "show all values" mode. From some basic digging, it appears uiInfo (line 457) in the simplemissionitem.cc is returning a null pointer. It appears to return a unique ID for PX4 and APM. Not entirely certain why this is occurring...

DonLakeFlyer, 2016-09-28:
> Waypoint icons are not displayed,

Are you building with a version of Qt greater than 5.5.1? QGC won't work with ny version of Qt greater than that. The symptoms of using it are things don't show up on the map.

The other problems are most likely caused by changes in MissionCommandTree which manages the editing on mission items. That went through a major change and hasn't had  all the bugs wrung out of it yet.

mattmccrink, 2016-09-28:
I am using QGC 5.5.1
The mission planner seems to work fine with the PX4 and APM stack.

DonLakeFlyer, 2016-09-28:
Hmm, strange. Likely MissionCommandTree changes causing both problems. This is not a protocol problem.

DonLakeFlyer, 2016-09-29:
I just did that:
- Built latest master
- Set offline editing to mavlink generic
- Rebooted
- Went to Plan, added a bunch of waypoints and they all showed up and the editors were correct

Are you sure you are running latest master? I fixed a bug similar to this problem last week.

mattmccrink, 2016-09-30:
I am using the latest master branch and am still having the same issue. Waypoint icons are not visible, and the "friendyview" is not accessible.
[image]

DonLakeFlyer, 2016-09-30:
When you set up offline editing to generic what vehicle type are you selecting. Not sure why I can't repro.

mattmccrink, 2016-10-03:
I have tried fixed wing and multirotor. Same effect.

DonLakeFlyer, 2016-10-11:
Can you turn on: MissionCommandsLog, https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html. And attach the log?

mattmccrink, 2016-10-12:
Here is the command log for a generic firmware editor. I attempted to create several waypoints to generate useful error messages. 
[MissionCommandLog.txt](https://github.com/mavlink/qgroundcontrol/files/523355/MissionCommandLog.txt)

DonLakeFlyer, 2016-10-13:
I assume you are building yourself. Have you modified the mission command json files? Look like the json file is not loading due to a badly formatted file:

```
[E] at :0 - "json document is not array"
```

I don't get this on my builds from master. Does a daily build from the installer work for you?

mattmccrink, 2016-10-14:
I just built the latest version and it seems to be working. Problem solved?

DonLakeFlyer, 2016-10-14:
Yes. I think you had an old master.

closed 2016-10-14 by DonLakeFlyer