mavlink/qgroundcontrol #4084: ReferenceError: shortcuts is not defined

bkueng, 2016-09-29:
QGC master crashes with
```
qrc:qml/QGroundControl/Controls/AppMessages.qml:116: ReferenceError: shortcuts is not defined
```
when trying to open the Settings -> Console view.

yzzir, 2016-12-14:
Also have this exact error with 3 different versions (AppImage, Compressed Archive, git clone build)

dogmaphobic, 2016-12-14:
What OS?

yzzir, 2016-12-15:
Sorry forgot to mention that.  Kubuntu 16.04

dogmaphobic, 2016-12-15:
I will try with Ubuntu and see if I can see anything weird. Strange that it would "crash" by just defining a hint for which directory to open the file picker.

yzzir, 2016-12-15:
One thing to mention, I install Qt using the opensource binaries from the Qt website, not from ubuntu's package management repos.

dogmaphobic, 2016-12-15:
I tried it on Ubuntu 16.04 and could not reproduce the problem. The file picker came up just fine pointing to my "home" directory as it should. Needless to say, Mac OS runs fine too.

Is there anything weird with your "user" account or your "home" directory?

bkueng, 2016-12-16:
I'm on Fedora, also KDE. It seems to be a bug in Qt, if I use the system's Qt (5.6.2), it works, but fails with Qt 5.5.1.

yzzir, 2016-12-16:
I was thinking that might be the case, as I am using Qt5.5.1 binary from qt.io on my Kubuntu 16.04 laptop which doesn't work, but it works on my Ubuntu 16.04 desktop when using system qt

DonLakeFlyer, 2016-12-22:
Likely some missing package on the version of Qt you are using. Hence the readme pushing folks to not use the built-in version but to install a Qt using the online installer. Looks like there are more reason than just what the README says:

```
You need to install Qt as described below instead of using pre-built packages from say, a Linux distribution because QGroundControl needs access to private Qt headers.
```

If you follow the README instructions do things work correctly?

closed 2023-08-24 by bkueng