mavlink/qgroundcontrol #4085: Question about pressure data

JM-SNU, 2016-09-29:
Hello, all.

I have some problems with obtaining pressure information from pixhawk.

I use qgroundcontrol software to set "pixhawk", and receive sensor data (e.g. gyro, acc, baro, ... ) using "mavros" package in ROS.

When I use 2.7.1 version, I can receive air pressure value from pixhawk (ROS topic is mavros/imu/atm_pressure).

However, after updating firmware and setting pixhawk using 3.0.1 version, the air pressure value is always zero.

That means, I can subscribe topic about air pressure but cannot receive correct pressure data.

I'm not sure, but I think it is problem with version of qgroundcontrol. (or parameter setting in updated version).

If you know how I can handle this problem, let me know.

Thanks.

DonLakeFlyer, 2016-09-30:
Not sure how QGC comes into play here? I assume air pressure comes across in a mavlink message from the vehicle. QGC doesn't have any control over that. If the message isn't coming across then something is wrong on the vehicle side. Possibly some setup issue. You can yuse Widgets / Mavlink Inspector to see that mavlink messages that are coming across.

booo, 2022-11-01:
No response after initial answer to question therefore I close the question.

closed 2022-11-01 by booo