mavlink/qgroundcontrol #4088: Unknown module(s) in QT: svg multimedia serialport

RickBlacker, 2016-09-30:
Hi all.

I downloaded the source. I already had QT installed. Opened up the project, tried to build. I'm getting the error 
**Unknown module(s) in QT: svg multimedia serialport**
Was hoping someone might be able to help me out here.  I searched the net but didn't find anything that was specific to this error.

Thanks for any help.
Rick

DonLakeFlyer, 2016-09-30:
From the README:

You need to install Qt as described below instead of using pre-built packages from say, a Linux distribution because QGroundControl needs access to private Qt headers.

I'll re-open if it's a real problem.

closed 2016-09-30 by DonLakeFlyer

RickBlacker, 2016-09-30:
So we can ONLY use QT 5.5.1?

DonLakeFlyer, 2016-09-30:
At the moment that is correct.