mavlink/qgroundcontrol #4089: size of array '_GStaticAssertCompileTimeAssertion_0' is negative

RickBlacker, 2016-09-30:
Opened up QT 5.5.1
Opened up the qgroundcontrol project file
Tried to compile and getting this error

/usr/include/glib-2.0/glib/gmacros.h:142: error: size of array '_GStaticAssertCompileTimeAssertion_0' is negative  __attribute__((__unused__))

This is coming from gmacros.h

Hoping to get a little assistance.  Is there a specific version of glib that's required?

Thanks
Rick

DonLakeFlyer, 2016-09-30:
I'd ask on the Gitter channel. You'll hit more Linux people there. If you find out, then repost here and we'll update README.

arno01, 2018-12-31:
Since google brought me here:

> ``/usr/lib/glib-2.0/include`` is, generally, platform-specific. It probably contains **32-bit** specific definitions. e.g., I have the following in ``/usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h``:
>
> ```#define GLIB_SIZEOF_SIZE_T 8```
>
> That would probably be ``4`` in your version, since it seems to be the **32-bit** one.

Source: https://stackoverflow.com/a/24541542/7246020

DonLakeFlyer, 2018-12-31:
Are you having compile problems? This issue is pretty ancient.

closed 2024-06-01 by HTRamsey