mavlink/qgroundcontrol #4092: Parameter editor improvements

AndKe, 2016-10-01:
As-is, the parameter editor is hard to work with, especially when tuning or really repetitively editing. 
Imagine editing , clicking a wrong parameter,requires "cancel" click, verifying that you clicked/editing correct parameter, needs an extra glance, after save, if parameter is to be changed, needs to be re-located in the list, then clicked.

Please follow me on this suggestion for an experience: (ignore my colors, it's just for illustration)
"Tools" button is always visible, maybe to the right at "search" level.

-User clicks RATE_RLL_D (in parameter list)
-the RATE_RLL_D line is highlighted
-right editing pane opens 
(we don't need to say "parameter editor"  , let's label it clearly with what we are editing)
(if user is editing in a tablet, it's a pain in the back to place cursor,change numbers, it's a constant danger of messing it up - as the editor limits/requires a number of digits, and some often need to be deleted/replaced, - giving a risk of decimal error. (or an zero missing))
-so - user can click +/- 10 or 50% buttons to change values safely
(changing 0 to 1 as text is trivial anyway, but using buttons on floating numbers is safer)
-every changed parameter changes color in the full paramter list, (and maybe show a count of changed values)
-now we *write* - (let's use same terminology as others, load/save is file, read/write is parameter.
- the "editor" does *not* disappear or close, it remain open in case we wish to further tune the values.
-there is a "read" button, that restores the current value in the text-box.

(so user was about to change something - pressed read - the box shows current value)
- want to change another parameter ? - at any time, just click it in the list . (no need to cancel first)

-now the great feature for those who tune:
history: every parameter, is in the history as long application runs
history starts with whatever a parameter is, at the time the parameters are fetched, then the list grows with every written parameter.

an user tuning in flight, or setting up second flaps output on ground, will always see what was already tried - very useful (replaces pen+paper :))

Finally : keyboard shortcuts Ctrl-R for "read" (current paramter) and Ctrl-W (all edited parameters) are essential as it allows the user to remain in the textbox, and quickly enter/change a value, hit hotkey, and don't even touch the mouse while working on a setting. 

[image]

DonLakeFlyer, 2016-10-07:
I think a better solution for this is to creating an actual "PID tuning" page which allows you to change all the PIDS in one place without using a side dialog. Also QGC needs increment/decrement editing support to make tablet editing easier.

NaterGator, 2016-10-07:
A flight script would be nice to. You would preplan a set of gains in a group, press one button to set the lot of them, could have multiple gain sets preplanned. If something goes wrong with the set, a big fat "undo" button to back out all the changes you just made.

AndKe, 2016-10-07:
@NaterGator different PID sets are best achieved already by loading a reduced .param file. (or - not in QGC yet ?  - do a compare, then select values for import)

@DonLakeFlyer Yes, a tuning page would do the trick to, the "new" thing I suggest is that user can see previously tried values.   But again, it does not only for PID's  - when setting secnod flap throw on a bigger plane user goes to and from deflection measuring instrument, the wing, and PWM entry in RCx_MIN/MAX - even there it would be very valuable to see what you tried before, it speeds up the adjusting process.

ChristophTobler, 2017-01-25:
Setting the focus to the input field would already make things much better. We would really appreciate that for parameter tuning in Switzerland where it is -8 degrees Celsius (17,6 degrees Fahrenheit). Every second where you need to take off your gloves and use the trackpad is a real pain ;)

dagar, 2017-01-25:
Closer? https://github.com/mavlink/qgroundcontrol/pull/4458

ChristophTobler, 2017-02-01:
Yes :+1: Looks cool. I'll give this a try

closed 2026-10-04 by github-actions[bot]