mavlink/qgroundcontrol #4093: Survey: mission text and droplist box size/text crop

AndKe, 2016-10-01:
Now all mission items are trimmed like this: (tested on two Ubuntu systems)
[image]

this anomaly is not seen QGC settings, parameters, or any other place than mission items.
all text boxes and droplist boxes(camera list) are affected. 
Only in the waypoint view, when changing a waypoint type -using the "category" droplist is ok. (the droplist with items like "basic, camera, advanced")

This issue seems to be introduced with https://github.com/mavlink/qgroundcontrol/issues/4087

AndKe, 2016-11-15:
If anyone has any clue as to what causes this, (a property of box or group?)  - I'll be happy to try.

DonLakeFlyer, 2016-11-15:
Is this on Windows?

DonLakeFlyer, 2016-11-15:
Oops, sorry. On Ubunut. Hmm.

DonLakeFlyer, 2016-11-15:
The strange combo box font thing seems to happen more on Windows.

AndKe, 2016-11-15:
It's important to mention it's not application-wide problem - but limited to "mission items" only.

AndKe, 2016-11-15:
another clue: something is different between mission&fence vs rally points: 
[image]
[image]

DonLakeFlyer, 2016-12-24:
@AndKe Can you try this again on a daily build and see if it is still there?

AndKe, 2016-12-25:
It's fine now.  - which commit fixed it ?

DonLakeFlyer, 2016-12-25:
#4310

closed 2016-12-25 by DonLakeFlyer

AndKe, 2019-01-07:
@DonLakeFlyer 
Please apply thesame fix(es) to current master, 

[image]

"mission start" item has this issue
"set servo" , "control mount" , and many others, as well as some fields of waypoint.
[image]