mavlink/qgroundcontrol #4095: Compass Calibration Long Stating Issue

jcooper21, 2016-10-02:
With PX4 and a Zubax GPS it is impossible to have the flight control yawed and the compass facing straight.  If the fllight controller is yawed the compass has to be yawed in the same direction for a proper calibration and flight.

DonLakeFlyer, 2016-10-02:
Not sure what QGC is supposed to do about this?

jcooper21, 2016-10-02:
Isnt it a QGC calibration setup issue?

DonLakeFlyer, 2016-10-02:
Supported orientations come from the firmware. You can set orientations for both the main board and any external compasses (parameters).  All sensor calibration is implemented in the firmware. Not sure how QGC is in control of any of this?

jcooper21, 2016-10-02:
So this would be a PX4 issue then. ill open an issue there.

closed 2016-10-02 by jcooper21

DonLakeFlyer, 2016-10-02:
I think so. Is there a missing orientation you need?

jcooper21, 2016-10-02:
No thats all good