mavlink/qgroundcontrol #4098: BT connection with Quanum BT telemetry box

hackra76, 2016-10-03:
Hi,
I am using this BlueTooth telemetry module from Quanum: http://www.hobbyking.com/hobbyking/store/__82895__Quanum_Bluetooth_Telemetry_Box_for_433MHz_Radio_Modules_V_2_.html

I am unable to connect from Android QGroundControl app to this BT module. It works flawless with Tower, but I like QGC much better... Is it possible to get it working ? Btw., it is a 433MHz module, connected to Pixhawk FC on a quadcopter. 
In Setup/Comm Links I added new Link by pressing Add. In the next dialog I chose a name, type as Bluetooth. Then press scan. I can see my Quanum BT module, so I select it and press OK. New Comm Link is created. I select it and press Connect. I get a yellow rectangle (error message): "Connection to service failed", nothing else. The blue LED on the Quanum BT module is still blinking, meaning not connected. In Tower, after I select BT and press Connect, it immediately turns solid Blue and connects the telemetry.

DonLakeFlyer, 2016-10-03:
Dup of #3583. Still haven't had time to get to this yet. But I finally have  BT module to test with. So I'll get to it.

closed 2016-10-03 by DonLakeFlyer

hackra76, 2016-10-03:
Daniel Agar asked me per email to open new Issue.. Didn't seem like a dupe of iOS issue #3583... Sorry

DonLakeFlyer, 2016-10-03:
I've been using that issue to cover all bluetooth issues. I'll update it.