mavlink/qgroundcontrol #4101: Plane: Need icons for Mission/GeoFence/Rally buttons

DonLakeFlyer, 2016-10-03:


dogmaphobic, 2016-10-27:
I looked at it today and I think it would make more sense to make it into a tab control. If you miss click, and click somewhere in between the icons, you end up adding a waypoint (or action depending on the mode you are in).

So instead of this:
[image]

Use some tab like control (this is made up in Photoshop, not a real control):
[image]

DonLakeFlyer, 2016-11-08:
Tab won't work because you can have multiple of the little yellow boxes under it now just one. I'm think I'm going to just switch these to radio buttons instead of using a icon. Much simpler.

DonLakeFlyer, 2016-11-10:
Switched to Radio Buttons. No icons needed.

closed 2016-11-10 by DonLakeFlyer