mavlink/qgroundcontrol #4105: QGC is not opening

anujkumargupta, 2016-10-04:
Hi i am working ubuntu 14.04 64 bit system i downloaded the daily buid of qgc from its site and isntalled it and also install the additional packages given in github site. But the problem is whenever i start my QGc application is starts with full white background 
[image]

narendrakc-path, 2016-10-04:
The error log is 

anuj@anuj-vm:~$ cd Downloads/qgroundcontrol/
anuj@anuj-vm:~/Downloads/qgroundcontrol$ ./qgroundcontrol-start.sh 
Settings location "/home/anuj/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"

(gst-plugin-scanner:2431): GStreamer-WARNING **: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgsteglglessink.so': /usr/lib/x86_64-linux-gnu/libgstegl-1.0.so.0: undefined symbol: eglDestroyImageKHR
Map Cache in: "/home/anuj/.cache/QGCMapCache300" / "qgcMapCache.db"
qrc:qml/MainWindowInner.qml:311:5: Type FlightDisplayView unavailable
qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayView.qml:17:1: plugin cannot be loaded for module "QtMultimedia": Cannot load library /home/anuj/Downloads/qgroundcontrol/Qt/qml/QtMultimedia/libdeclarative_multimedia.so: (libQt5MultimediaQuick_p.so.5: cannot open shared object file: No such file or directory)
Orphaned log file count 0

Thanks

DonLakeFlyer, 2016-10-04:
@dagar Any ideas?

dagar, 2016-10-04:
I guess libQt5MultimediaQuick_p.so.5 is a new dependency.

anujkumargupta, 2016-10-07:
Hi I tried to clone the code and build using qt5.5.1 and now it works properly 
Thanks

dagar, 2016-10-07:
I'll still fix this for the Linux users that don't want to compile.

CarlOlsson, 2016-10-09:
Hi I have the same problem on Ubuntu 14.04, if I download the daily build it doesn't start but if I build from source it works.
[image]

DonLakeFlyer, 2016-10-10:
@dagar Any update on the fix for this?

dagar, 2016-10-11:
Working on a fix.

sanderux, 2016-10-12:
@DonLakeFlyer is this mergeable?

DonLakeFlyer, 2016-10-12:
@dagar Should be able to merge when he's ready. Not sure.

closed 2016-10-12 by dagar

anujkumargupta, 2016-10-19:
Hi 
today i downloaded the latest build but again it asking for some shared library
/home/anuj/Downloads/qgroundcontrol/QGroundControl: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory
I tried to locate it also but its not there in my pc
Thanks

dagar, 2016-10-19:
Ah, new problem already. New issue here - https://github.com/mavlink/qgroundcontrol/issues/4148
I should really add something in the builds to test running the AppImage.

misoragod, 2017-03-19:
Hi.
I have the similar problem on Ubuntu 14.0.4 when I built from source to use Qt Creator.
I used Qt 5.5.1 and Qt Creator 4.2
[image]

DonLakeFlyer, 2017-03-19:
Use Qt 5.7.1

misoragod, 2017-03-20:
Hi.
I tried it to use Qt 5.7.1 and then I could solve this problem.
Thanks.

PR #4122 (2016-10-12): linux package always include libQt5Multimedia
 - fixes #4105 

My default qmake build had QGC_DISABLE_UVC defined, but still needed these libs. Any objection to just leaving them in all the time?
commit 8861c1f7 (2016-10-12): linux package always include libQt5Multimedia (#4122)

- fixes #4105