mavlink/qgroundcontrol #4112: Disable UVC video input by default

LorenzMeier, 2016-10-06:
QGC tries to open a video device (*NOT TO BE CONFUSED WITH DIGITAL VIDEO STREAMS*) by default and segfaults on a number of platforms (I believe we had issues on Android and iOS). I think we should disable this by default and let users enable it optionally.

Either QGC or Qt is very unstable on this feature and unless you have an analog-to-digital video USB stick connected it makes no sense for most people.

LorenzMeier, 2016-10-06:
@bkueng This is to track the crash that you were experiencing

dogmaphobic, 2016-10-06:
Qt crashes when "looking" to see if there is a camera. This is on Linux only. It works fine in all other platforms. This feature is disabled by default on Linux.

https://github.com/mavlink/qgroundcontrol/blob/master/qgroundcontrol.pro#L56-L61

LorenzMeier, 2016-10-06:
I'm telling you it crashes on that location on Android.

LorenzMeier, 2016-10-06:
Also, the whole logic seems backwards - why are you setting a define to disable something? Shouldn't it be enabled on each platform where we know its stable?

dogmaphobic, 2016-10-06:
> I'm telling you it crashes on that location on Android.

Android is Linux but there is no Android device with no Camera (that we use anyway).

> Also, the whole logic seems backwards - why are you setting a define to disable something? Shouldn't it be enabled on each platform where we know its stable?

It is stable and it has been fixed in newer versions of Qt. We're stuck with 5.5.1 so for the time being, I disabled it on Linux.

DonLakeFlyer, 2016-10-06:
Just FYI it doesn't crash on my Samsung tablet. In fact it shows the tablet camera in the little video box! Definitely need some user way to turn off.

bkueng, 2016-10-07:
Some more infos: QGC reproducibly crashes on startup, android with latest master. I don't even get to see the main widget.
http://pasted.co/1386a63d

LorenzMeier, 2016-10-07:
@bkueng Since we are using an x86 tablet the best initial step (given that x86 tablets are really weird on Android) would be to use this to exclude UVC there for now:
```
android {

    equals(ANDROID_TARGET_ARCH, armeabi-v7a) { 
        [doAndroidArmStuff]
    }
    equals(ANDROID_TARGET_ARCH, armeabi) {
        [doAndroidArmeabiStuff]
    }
    equals(ANDROID_TARGET_ARCH, x86)  {
        [doAndroidx86Stuff]
    }
}
```

bkueng, 2016-10-10:
Seems to work now, not sure what the problem was, but manually uninstalling QGC and flashing it resolved the problem.

booo, 2022-11-01:
Seems to be resolved.

closed 2022-11-01 by booo