mavlink/qgroundcontrol #4113: Figure out video streaming for iOS

DonLakeFlyer, 2016-10-07:


helmarw, 2016-12-16:
is there a solution already ?!
the video window is there and i set up video streaming on the rpi
(e.g. works with tower video widget on android)
but im missing the settings for iOS
thnx 
Helmar

dogmaphobic, 2016-12-16:
I've had video streaming on iOS since I first wrote the video streaming module. What is the issue?

helmarw, 2016-12-16:
nothing comes up (no video), and i cant find the settings for it either,
maybe im just blind ;) 

need to change the port to 5000 for the stream 

mavlins port is default 14500 

thnx  

Helmar

 

Links:
------
[1]
https://github.com/mavlink/qgroundcontrol/issues/4113#issuecomment-267665823
[2]
https://github.com/

dogmaphobic, 2016-12-16:
I can't find my RPI scripts. I haven't used the RPI in months but nothing changed since. Note that for RTP, the UDP port is 5600. I had a Logitech camera connected to the RPI using USB. It would show up as a normal /dev/video*n* device. The script I had was not that different from the one described here:
```
gst-launch-1.0 uvch264src initial-bitrate=1000000 average-bitrate=1000000 iframe-period=1000 device=/dev/video0 name=src auto-start=true src.vidsrc ! video/x-h264,width=1920,height=1080,framerate=24/1 ! h264parse ! rtph264pay ! udpsink host=xxx.xxx.xxx.xxx port=5600
```
Where xxx.xxx.xxx.xxx is the iOS device IP address.

The changes I had were related to my camera showing up as /dev/video1 instead of video0 and I used 720P instead of 1080P.

Make sure the iOS device is connected to the same network as the RPI. I had the RPI as the AP and configured to provide DHCP services but a fixed IP address to the iOS device's MAC address where QGC was running.

Also make sure you have the proper video source selected along with the proper UDP port for RTP video streaming.

[image]

helmarw, 2016-12-16:
i dont have the option for video under general... talking about iOS not macOS

dogmaphobic, 2016-12-16:
I took a screen grab off Mac OS because it was simpler. Now I see why Don created this issue. It's to see how to enable video streaming from *official* builds. I guess it's not yet enabled and if that's the case, the only way to enable for now is to build your own.

[image]

helmarw, 2016-12-16:
well, neither in macOS nor iOS i got the video option ...

 

Links:
------
[1]
https://cloud.githubusercontent.com/assets/749243/21276149/1c3fdaf2-c39e-11e6-81f6-51fffa95f868.png
[2]
https://github.com/mavlink/qgroundcontrol/issues/4113#issuecomment-267680760
[3]
https://github.com/

helmarw, 2016-12-16:
ok, that explains it, thnx

 

Links:
------
[1]
https://cloud.githubusercontent.com/assets/749243/21277499/a7efb3e6-c3a4-11e6-8fb7-2e008e8473bc.PNG
[2]
https://github.com/mavlink/qgroundcontrol/issues/4113#issuecomment-267689117
[3]
https://github.com/

DonLakeFlyer, 2016-12-22:
@dogmaphobic Would it be possible to get this into official builds for 3.1?

dogmaphobic, 2016-12-22:
Technically it's just a matter of adding the GStreamer libraries to the build system. Based on what I see in the CI build scrips, the only OS that doesn't include it is iOS. I checked the lasted CI logs and indeed it shows Mac OS building with video streaming support, which makes @helmarw assertion above (that he has no video options) strange.

DonLakeFlyer, 2017-05-25:
Too late for 3.2

DonLakeFlyer, 2017-05-25:
Would be good to get this turned on for beginning of 3.3 so it can have good bake time.

DonLakeFlyer, 2017-12-29:
@dogmaphobic Will we be able to get video streaming for Stable 3.3 builds?

dogmaphobic, 2017-12-30:
Video streaming has been working on iOS since the beginning :)

[image]

closed 2017-12-30 by DonLakeFlyer