mavlink/qgroundcontrol #4119: Add a preflight check status pane

LorenzMeier, 2016-10-08:
In contrast to the filter status pane (which is technical) the preflight status pane should be focused entirely on operation and might be visible by default on the flight view as overlay while not armed.

http://mavlink.org/messages/common#SYS_STATUS

philipoe, 2018-03-01:
Is anyone working on this at the moment? 

We will be needing this rather soon, I can offer to work on this a bit. The overall idea is to implement a pre-flight checklist as many companies have done already. For example, see this variant by Wingtra(image retrieved from the [wingtra website](www.wingtra.com). Also see their video at  https://www.youtube.com/embed/S-MfYFBHnr8 )

[image]

I have a list of checklist items in place. There should be a set of base items (e.g. IMU OK, estimator OK, ...) that are checked for every platform and some reconfigurable items (like SatCom OK, Camera started). We can discuss more later. For now, we should define/discuss:

1. Where in QGC should this go? As Lorenz indicated, I'd suggest an overlay in the plan view that stays there until the mission is started.
2. Can this be coded in QML or not? I still need to get up-to-speed in QML, but would assume that due to some of the limitations of QML and the fact that the checklist is a _core_ component of QGC, it cannot be fully implemented with QML only. For example, accessing certain non-common mavlink message set definitions is impossible. Whats your input on this?
3. How can one make sure that this works for both fixed-wing and multicopters? I assume the combination of standard+reconfigurable check list items is required for that.

@LorenzMeier @DonLakeFlyer

LorenzMeier, 2018-03-04:
@dogmaphobic @bkueng @dagar This is related to the discussion we had. @philipoe @dogmaphobic Could you sync through slack (http://slack.px4.io) and arrange a call to discuss?

dogmaphobic, 2018-03-04:
This should be integrated with Airspace management as it is also part of the check list. Otherwise we will end up with two, separate "check lists".

dagar, 2018-03-04:
Very interested in this! Related Firmware issue. https://github.com/PX4/Firmware/issues/8697

My initial suggestion is to drive as much of this as possible from the vehicle side. Let's expose the actual flags PX4 is using for the various checks. In cases where user input is required to effectively check something the vehicle should guide the user through appropriate actions.

We could have various levels of this, with the full set being forced for a new vehicle setup where things are very prone to being wrong.

Other ideas (optional early checks for new setups)
 - vehicle could walk through moving each control surface to each extreme (and center) and ask the user to verify. The vehicle can command each control surface independently, a pilot can't.
 - ask the user to point the vehicle North, then rotate (mag issues are still a huge pain point)
 - automated checks? A multicopter could throttle up a single motor at a time to verify the proper response. Checks motor connection, numbering, direction, etc.

Antiheavy, 2018-03-05:
I'm also very interested.   I posted some related thoughts in the firmware issue @dagar linked above (https://github.com/PX4/Firmware/issues/8697).  Here are some more thoughts:

There are two concepts to consider:  "Preflight status" and "Preflight checklist".  They are different concepts, but not mutually exclusive.  The better automated Preflight checklists I've used did a good job of embedding "status" into the checklist and UI in general.  

I'm not a huge fan of this "checklist" or "status" living in the Plan screen, but then again I'm not a fan of the Plan screen in general (Fly and Plan should be unified, but that is a different discussion).

Instead, this should be an overlay or a pop-up on the main Fly screen.  Activated by a "Preflight checklist" button at the bottom of the screen (in place of the "Start Mission" slider).  "Preflight checklist" must be all checked green before "Start Mission" slider is allowed to be shown.  If the automated status of any items change before arming, "Start Mission" should be hidden and "Preflight checklist" should pop-up with the offending items shown red.

It is impossible to have one generic preflight checklist that will work with all platforms.  Also, there will always be items on the preflight checklist that the autopilot has no way of knowing about.  The goal here should be to make a flexible "checklist" UI than can adapted to different platforms.   The checklist UI should be take "status" info from the autopilot as much as possible.

When customizing a preflight checklist, a design goal should make it as short and efficient as possible while ensuring all critical items get checked.  I've seen both "lists" (like the example shown above) and "wizards" (...next -> next-> next...) work well in past systems I've used.  Pros and cons to both.  I feel people don't use "wizard" style interfaces much anymore for whatever reason.

Here is a screengrab of one of our fixed-wing preflight checklists (manual checklist, hard copy/PDF):
[image]

philipoe, 2018-04-11:
Quick update, given that I have been working on this on both the autopilot (Pixhawk) and ground station side:

### GUI overview GCS side
The GUI below is the current setup for a fixed-wing vehicle. Directly after startup, not many tests run yet:
[image]

A bit further into startup for indoor and outdoor scheme with some automatic and some manual checks passed already (automatic battery check fails because i am running on USB):

[image]
[image]

### Overview
 - Designed to use as many automatic checks as possible. In cooperation with @dagar, we've been working on getting the MAV_SYS_SENSOR_STATUS flags (https://mavlink.io/en/messages/common.html#MAV_SYS_STATUS_SENSOR) in the SYS_STATUS message https://mavlink.io/en/messages/common.html#SYS_STATUS filled with something meaningful in PX4 (currently still on our local repositories). This mainly means that commander & preflight check are automatically checking for connection/calibration/validity of a certain sensor (currently implemented: Accel, Mag, Gyro, Baro, Airspeed, GPS, RC). This means that, as indicated in some comments above, the checks are driven from the vehicle side, and QGC just has to check the validity flags.
 - Some manual or hybrid checks remain. These can be multi-step tests: For example, the RC connection and calibration of the RC are currently checked automatically, and when this is OK, the respective check list item changes to a "Perform range test and confirm".
 - This is also NOT targeted towards the hobbyist "just quickly flying his quadcopter in his garden", but towards slightly more serious applications of UAVs.
 - This is NOT supposed to just indicate system status (QGC has other means for that), but really IS designed to replace your standard paper-based preflight checklist. I know there exist more concise versions of a checklist or check-method (e.g. in DJI vehicles), but we have made the experience that with non-COTS/modified platforms there are always custom or not-automatically-checkable items ("are the screws in part XXY inserted") that one still definitely needs to check. In our case, this semi-automated checklist will for example replace our current 2.5 page paper-based checklist.
 - This should in the end be reconfigurable. As a function of the airframe set in QGC, different .xml files (fixed-wing, multicopter, ground rover...) can be loaded. If a user wants to add specific checklist items ("is my super-truper 4th camera OK?"), then this could also be allowed via a custom.xml file or so.
 - At some point, we could connect the checklist such that the "start mission" button only pops up if the checklist is completed. However, one will also be able to completely disable the checklist via the settings (for pro-users).

### Questions:
 - GUI ok? It's currently an item in this drop tools menu of the flight view.
 - Any other feedback?

@dagar @DonLakeFlyer Happy about your input.

DonLakeFlyer, 2018-04-11:
My one first suggestion here is to not wait until everything is completedly done to put through your first pull request on this. It's much better to put through smaller pull requests with incremental updates. This makes it much easier to code review instead of just dumping one massive change all at once.

DonLakeFlyer, 2018-04-11:
You can add a setting to AppSettings to control this, then add a checkbox to General settings page to turn on/off.

Antiheavy, 2018-04-12:
> This should in the end be reconfigurable. As a function of the airframe set in QGC, different .xml files

yes!  awesome.

> "start mission" button only pops up if the checklist is completed.

Yes!

> GUI ok? It's currently an item in this drop tools menu of the flight view.

I like that you've put it in the Fly toolbar.  I can't think of a better place for it right now.  Should it have some sort of animation and/or color coding to indicate it needs user action to complete.  The animation/colors should activate once connected to a vehicle and again when checklist and status is 100% complete and good.  I haven't thought enough of specific suggestions here.

DonLakeFlyer, 2018-04-12:
> This should in the end be reconfigurable.

To me, that is going to be the most important and difficult part of this whole thing. Not knowing the vehicle type and picking the right list. But being able to set up your own pre-flight checklist items.

dagar, 2018-04-12:
> > This should in the end be reconfigurable.
> 
> To me, that is going to be the most important and difficult part of this whole thing. Not knowing the vehicle type and picking the right list. But being able to set up your own pre-flight checklist items.

There's no avoiding the vehicle specific manual steps, but this is why I'd like to make sure we can push as much of this as possible to the vehicle itself. In many cases the vehicle can tell us both the status of something (eg RC, GPS, data link) and if it should actually be required (without digging into parameters) for operation (my interpretation of onboard_control_sensors_enabled).  

[image]

[image]

closed 2019-10-03 by LorenzMeier