mavlink/qgroundcontrol #4121: Iridium support

dagar, 2016-10-11:
I'm working on adding a single mavlink message appropriate for Iridium connections. https://github.com/mavlink/mavlink/pull/634

On the QGC implementing the new message is easy, but what about things like connection timeout (HEARTBEAT?), and preventing things like mission and param sync that just wouldn't work.

PX4 side - https://github.com/PX4/Firmware/pull/5633

DonLakeFlyer, 2016-10-11:
Can you explain what this is? Is it a single direction really slow link?

dagar, 2016-10-11:
It's a satellite connection, think 50 bytes getting through once every few minutes. Not single direction.

In my particular situation I'll connect with a regular SiK radio first, sync a mission, and takeoff. Then over Iridium I'm mostly just monitoring status, and possibly sending single commands.

fvantienen, 2016-10-12:
@dagar Take a look at https://github.com/tudelft/iridium
This is the iridium datalink we used during the outback Medical Express Challenge. And it only has on average 2 seconds delay (4 seconds round time) and a maximum data rate of 9600bps.

closed 2026-10-04 by github-actions[bot]