mavlink/qgroundcontrol #4124: Please could you display correctly custom commands

Fojtik, 2016-10-11:
Custom commands are not displayed correctly in the version 3.0.1.

Please see this:
[https://github.com/mavlink/mavlink/issues/525](https://github.com/mavlink/mavlink/issues/525)

DonLakeFlyer, 2016-10-12:
Working on it. Turned out to be more difficult than I thought it would be. Was hoping to get it done today.

Fojtik, 2016-10-12:
Perfect, thank you very much for this feature. I can test it, when it will be finished.

Fojtik, 2016-10-13:
I do not know whether you finished it:
[image] Unfortunatelly circle position is not driven by Latitude and Longitude.
Is it possible to fix it.

DonLakeFlyer, 2016-12-27:
Going to have to wait till 3.2 since it's a large change now. Sorry for taking so long.

Fojtik, 2016-12-30:
OK, thank you very much for your interest. Please let me know when these points will work - I can test it.

DonLakeFlyer, 2017-05-26:
I screwed this up again in thinking it was easy change but turned out it was not. And then it got lost until too late for 3.2. I have a new system for marking fixes which needs to go in at the start of a new release so this won't happen again. The Issue tracking this is #3116

closed 2017-05-26 by DonLakeFlyer