mavlink/qgroundcontrol #4126: Accelerometer calibration problem

marstonstudio, 2016-10-13:
I am unable to complete the Accelerometer calibration using QGroundControl. 

When I complete the orientation steps, I see a message in the console 'Info: Calibration Succesful'. When I try to arm it, I see a message 'Critical: PreArm: Accelerometers calibrated requires reboot.'

However after rebooting my system, when I reconnect QGroundControl, the accelerometer is still red and there is no record of the calibration.

Is there anything special I need to do to save the values before rebooting my copter?

I am running APM:Copter V3.4-rc4 (9ea82ab9) on a Navio2 with a Raspberry Pi 3. I am able to complete the calibration successfully using Mission Planner on Windows, but I'd rather use QGroundControl on my Mac.

Happy to submit logs or any other debugging info if this is a bug. Grateful for any guidance if I'm missing a step.

( previously posted in google group: https://groups.google.com/forum/#!topic/qgroundcontrol/E_eQkDfIZhQ )

DonLakeFlyer, 2016-10-20:
@marstonstudio  Can you go through the sequence of steps again which fails. Then after you reboot and Sensors still shows red. Save your paramateres to a file on post here.

marstonstudio, 2016-10-20:
One question, what is the best way to save the parameters to a file?

marstonstudio, 2016-10-21:
Attached are parameter files from before calibration, after calibration, and after reboot. I have also attached a screenshot showing the values that I got after completing the accelerometer calibration.

[image]

[aftercalibration.params.txt](https://github.com/mavlink/qgroundcontrol/files/543310/aftercalibration.params.txt)
[afterreboot.params.txt](https://github.com/mavlink/qgroundcontrol/files/543311/afterreboot.params.txt)
[beforecalibration.params.txt](https://github.com/mavlink/qgroundcontrol/files/543312/beforecalibration.params.txt)

DonLakeFlyer, 2016-11-10:
Perfect, I understand the problem now. Working on a fix. The bug is that the indiacator is incorrect since you only have one accelerometer. Workaround is to set INS_USE2 and INS_USE3 to off. Then the indicator should go green.

Deathstroke5467, 2016-11-17:
Sorry I was looking at this myself and hadn't been working for me either. Though the "Navio2 contains two 9DOF IMU - MPU9250 and LSM9DS1" as found in the official Navio 2 Docs. So I'm not sure what you mean by theres only one accelerometer?

staroselskii, 2016-11-18:
@Deathstroke5467 

Navio 2 has only one IMU. LSM9DS1 is used merely as a compass.

closed 2016-11-18 by DonLakeFlyer