mavlink/qgroundcontrol #4132: Switch to Qt 5.6.2

DonLakeFlyer, 2016-10-14:
Now available:
* Fixes lines on map problem
* Allows for @gregd72002 joystick pull to go through

yeechongyeung, 2016-10-15:
Is QGC can works with Qt5.6.2 ?

DonLakeFlyer, 2016-10-15:
Initial testing says yes. But hasn't been fully run through yet.

nanthony21, 2016-10-16:
I'd like to get this working. What kind of testing needs to be done?

DonLakeFlyer, 2016-10-17:
Need to run through basic manual tests on all platforms/devices:
* Connect
* Create missions
* Test flights through SITL

Also need to update travis Qt install packages. For now I think I'm going to push this to 3.2 to give it more daily build bake time. I'm pretty close to closing down 3.1 for a new point release.

nanthony21, 2016-10-18:
Tested a little bit on Linux today. Only thing I noticed was that after drawing a polygon for a "Survey" mission item if you then click "Draw" again the blue polygon and green line will appear exactly as they were when you ended the first drawing. Once you begin drawing the second polygon they go away.

nanthony21, 2016-10-21:
I've done the testing mentioned above on Ubuntu and Android and haven't noticed any problems. The only thing I haven't tried is SITL on android, not sure how to do it or if it's possible.

gregd72002, 2016-10-31:
I have refreshed the PR for Android Joystick (requires >Qt 5.6). It looks like travis needs adjusting as it still uses Qt 5.5.1 for builds.

https://github.com/mavlink/qgroundcontrol/pull/4179

nanthony21, 2016-11-04:
I notice that there is only one bug left in the QT 5.7.1 bug tracker. Does that mean that we may jump straight the 5.7.1?

DonLakeFlyer, 2016-11-15:
5.6.2 doesn't fix any of the map bugs that cause problems (although it does show waypoint  lines). So this will wait to 3.2 since there isn't advantage to rushing this.

NaterGator, 2016-12-15:
@DonLakeFlyer [Qt 5.7.1](http://blog.qt.io/blog/2016/12/14/qt-5-7-1-released/) was released yesterday. I have been dogfooding the branch for a while and have been pleased. Keep in mind 5.7 include the GPLv3 QtCharts module which we could certainly use to improve the analysis views.

dogmaphobic, 2016-12-15:
I would wait for 5.8.x. 5.7 is a development branch. 5.8 is the next LTS release. Also, the charting widgets are pretty worthless for what we want. We're better off rolling our own, which I've spent some time toying with it already.

dogmaphobic, 2016-12-15:
This is a different project I've been working on. I needed to plot some graphs. I tried using the chart widgets but it's close to impossible to have it look the way you want. They work fine for "business" type plots but not for something custom. It was a lot simpler to create my own, custom chart control.

[image]

DonLakeFlyer, 2016-12-22:
Our charting needs are fairly simples:
1) Draw some lines
2) Good multiple axis drawing support

DonLakeFlyer, 2016-12-22:
I never realized that the even numbers where LTS releases till Gus told me a while ago. That said we've been on 5.5 (another non-LTS release) for a while now and are pretty far behind. Is there a timeline for 5.8? I think we release 3.1 (which is almost ready) and then move to a new Qt as soon as that goes out.

NaterGator, 2016-12-22:
Yes, the RC was [released](http://blog.qt.io/blog/2016/12/22/qt-5-8-0-release-candidate-available/) today:
https://wiki.qt.io/Qt_5.8_Release

The final release is scheduled for mid-Jan, so that timing should work out well.

dogmaphobic, 2016-12-22:
> Our charting needs are fairly simples:

Yes, I spent some time going through the existing (Qwt) code trying to understand it. I'm still on the fence on how to go about it. One option is to use QtCharts, the other is to adapt Qwt so it can be used from QML. That's for the graphing part. The other half the headache will be to come up with a UI to define what to plot. The existing (QWidget) code is a nightmare (from an UX standpoint). I have yet to think of something that would work within the constraints of QML (and tiny displays). I even thought about having the realtime graphing running on a *slave* device (tablet or otherwise).

No matter how you look at this, Video Streaming and Mapping will look like a day in the park in comparison. I'm not there yet. I still need to think about this.

DonLakeFlyer, 2016-12-22:
I think Analyze is something that would not be available on small screens

dogmaphobic, 2016-12-22:
I'm not sure that's what @LorenzMeier has in mind though. The bulk of the headache is getting it usable on tiny screens (tablets).

DonLakeFlyer, 2016-12-22:
Tablets, yes. Phones no.

NaterGator, 2016-12-22:
The problem with QtCharts on mobile is the license. I'm not a lawyer but I don't see how the GPL3 QtCharts license can be compatible with the app store licensing requirements.

dogmaphobic, 2016-12-22:
> Tablets, yes. Phones no.

Anything smaller than 27" is "tiny" :P

> The problem with QtCharts on mobile is the license. I'm not a lawyer but I don't see how the GPL3 QtCharts license can be compatible with the app store licensing requirements.

I have a commercial license for Qt exactly for that reason. Otherwise we would not be able to upload it to the iOS app store.

dogmaphobic, 2016-12-22:
Basically, you are trying to fit this information into a (tiny) tablet screen. Take in consideration that we have a tool bar and a side menu taking quite a bit of real estate. That's why I thought about running it off a *slave* device.

[image]

NaterGator, 2017-01-24:
Qt 5.8 has been released: http://blog.qt.io/blog/2017/01/23/qt-5-8-released/

dogmaphobic, 2017-01-24:
Yes, I've been using the RC for a while. I just finished installing it on Windows to make sure it all works. It took a bit to get debugging going (you need to install the Windows SDK), but it all works.

anton-matosov, 2017-02-08:
I have start using 5.8 on OS X for mac and ios builds. Works pretty stable so far.

closed 2017-03-19 by DonLakeFlyer