mavlink/qgroundcontrol #4141: Wrong Firmware Version

nanthony21, 2016-10-16:
I'm not sure if this is a QGC problem or a firmware problem. When using QGC to flash the default stable PX4 firmware it says that it will use version 1.4.4 but the vehicle summary page shows that 1.4.1dev is installed.

dagar, 2016-10-16:
I'm not sure if this is a problem on the PX4 or QGC side, but I'll look into it.

dagar, 2016-10-20:
It's a problem with the current PX4 stable build. Not sure what yet.

dagar, 2016-10-20:
Seems like more of a PX4 build system issue.

dagar, 2016-10-20:
Check stable again once this build finishes - https://travis-ci.org/PX4/Firmware/builds/169102480

dagar, 2016-10-20:
It's partially fixed now, but displays 1.4.4dev because I added a commit after 1.4.4 to fix it. The next stable release will be correct.

closed 2016-10-20 by dagar