mavlink/qgroundcontrol #4148: Latest AppImage for linux needs libsdl2-dev

julianoes, 2016-10-18:
On Ubuntu 16.04, the latest daily AppImage (2015-10-18) gave the following error: 

```
/tmp/.mount_F2ihi7/QGroundControl: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory
```

This fixed it:
```
sudo apt install libsdl2-dev
```

I think this should be fixed in the guide here: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html

closed 2016-10-19 by dagar

dagar, 2016-10-19:
@julianoes @anujkumargupta - can you try the latest master? http://qgroundcontrol.s3.amazonaws.com/builds/master/QGroundControl.AppImage

julianoes, 2016-10-20:
Yes, seems to work without again, thanks.

PR #4155 (2016-10-19): AppImage update to libsdl2
 - fixes #4148
commit 1f20c604 (2016-10-19): AppImage update to libsdl2 (#4155)

- fixes #4148