mavlink/qgroundcontrol #4149: Mavlink inspector: many messages not recognised/displayed.

AndKe, 2016-10-18:
Please notice that the 180 (camera feedback) does not even appear.
many more are unrecognised. 

Also, displaying us as exponent is .... less than optimal.

[image]
[image]

DonLakeFlyer, 2016-10-18:
There is a bug in the mavlink 2 headers which prevents QGC from getting access to ardupilot message info. Once that is fixed I can pull new mavlink headers and this will start working again.

DonLakeFlyer, 2016-10-18:
https://github.com/mavlink/mavlink/issues/639

DonLakeFlyer, 2016-11-11:
Correct headers are in now

closed 2016-11-11 by DonLakeFlyer