mavlink/qgroundcontrol #4151: Issue on the parameter missing problem with Copter 3.2.1 firmware.

bestpest, 2016-10-19:
After connecting qgc to copter I see warning pop-up that parameters missing from firmware.

bestpest, 2016-10-19:
[image]

DonLakeFlyer, 2016-10-19:
There is another bug here as well. The ArduPilot branding logo should not show up if it overlaps the toolbar items.

DonLakeFlyer, 2016-10-20:
Installed Copter 3.2 and the COMPASS_DEV_ID parameters were there.

@bestpest Thought you said you were running Copter 3.2.1? Can you look on the Setup/Summary screen and check firmware version there?

bestpest, 2016-10-21:
[image]
[image]
[image]
[image]

DonLakeFlyer, 2016-11-24:
I think I just fixed this. ArduPilot was sending to fake parameter back with an index of 65536. This was causing some retry times to get out of whack.

closed 2016-11-24 by DonLakeFlyer

bikashranjanyadav, 2017-09-10:
I have two apm2.8 with two neo7m gps.
in both compass_dev_id  value is zero.( tried both copter 3.2.1 & beta version USING APM PLANNER 2.0)
when connected to qcgroundcontrol says compass_dev_id parameter missing
also, my compass heading is wrong after many attempts

WHY MY COMpASS IS NOT GETTING DETECTED pLZ HELp

DonLakeFlyer, 2017-09-10:
QGC no longer supports version 3.2. It's too old.