mavlink/qgroundcontrol #4154: "Vehicle did not respond to request for parameters" - a scare

AndKe, 2016-10-19:
So, seeing this scares user, maybe all were eventually fetched , maybe not . maybe opening a setup/config page will mess things up as current setting is assumed be "0" .

Please be more informative:
"Vehicle did not respond to parameter request, got 0 of (unknown) , retrying"
or, like the screenshot shows, some are fetched:
"Missing parameters, got 430 of 550, retrying."
"All parameters (550) received."

Is user allowed to use config screens before everything is loaded ?

[image]

DonLakeFlyer, 2016-10-19:
This message means what it says. The vehicle did not respond to the request for paramters. Until you get parameters the setup pages don't even show up so there is no way to screw things up. 

If you only get partial parameter list back you will get a different message:
```
        qgcApp()->showMessage("QGroundControl was unable to retrieve the full set of parameters from the vehicle. "
                              "This will cause QGroundControl to be unable to display its full user interface. "
                              "If you are using modified firmware, you may need to resolve any vehicle startup errors to resolve the issue. "
                              "If you are using standard firmware, you may need to upgrade to a newer version to resolve the issue.");
```

The console log will have details on what exact parameters are missing.

QGC usually says nothing it the parameter load succeeds. You just see the  progress bar go to the end of the screen. Probably makes sense for QGC to say "parameter load complete" or something if the user gets a "did not respond" message so they no eventually it all sorted itself out.

AndKe, 2016-10-19:
We basically agree; once some error is indicated, we should have something to confirm that all is well - if/when it is.   - or information if there is any progress.

DonLakeFlyer, 2016-11-24:
I think the solution is to toss the "Vehicle did not respond to request for parameters, retrying" message and just show the final state if there is an error. The user shouldn't care what is happening in the middle of the process, just if it fails completely.

AndKe, 2016-11-24:
Sounds good,. 
The full parameter view should make user aware when some parameters are missing;  (maybe similar to AP2 by displaying parameter count like "210/530")  

Finally,  (I did not check if it is like that already), tuning and config pages must be disabled, or at least do not write parameters that are missing.   (in order to prevent writing some params as 0 because they were not loaded.)

DonLakeFlyer, 2016-11-25:
New changes:
* Got rid "Vehicle did not respond to request for parameters, retrying" message
* Now you only get error messages when load process completes
* Param load progress bar is per-vehicle instead of global to all vehicles

Existing behavior still here
* Error messages tells you whether vehicle sent no params or whether it sent partial params (including how many are missing). This is not new, it already did this
* Setup screens don't show if no/partial param load. Parameter Editor does show in this case.

Remaining work for 3.2 release:
* Parameter Editor should show param load error details when you go there.