mavlink/qgroundcontrol #4163: HIL Quadrocopter X

lovettchris, 2016-10-20:
When I try and set "HIL Quadrocopter X" on a Pixhawk I always have to do it twice.  The first time seems to only set HILStar (XPlane), the second time "HIL Quadrocopter X" sticks.

DonLakeFlyer, 2016-11-14:
Going to need more exact details of the click you are making. It's working fine for me.

DonLakeFlyer, 2016-11-14:
Also if you are using a mouse-wheel to scroll up after you select it. Make sure the mouse-wheel isn't changing the selection in the dropdown.

DonLakeFlyer, 2016-11-24:
@lovettchris ?

clovett, 2016-11-24:
I can reproduce this easily.  I just selected "Generic Quadrotor X config" under "3DR DIY Quad" dropdown.  After reboot, AirFrame is set to  "3DR DIY Quad".  So I try again, this time it is correctly set to "Generic Quadrotor X config".  Now I try and set "HIL Quadrocopter X" and restart, but it is set to "HILStart (XPlane)".  Let me try and debug it for you...  I suspect there's a simple UI bug not noticing the correct combo dropdown selected index or something...

DonLakeFlyer, 2016-11-24:
What OS? Maybe some weird OS specific thing.

DonLakeFlyer, 2016-11-24:
Code is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/AirframeComponent.qml#L231

clovett, 2016-11-24:
Windows.  Thanks,  as soon as QTCreator has finished installing, I'll take a look.

DonLakeFlyer, 2016-11-24:
In staring at the code it may be a problem with selecting from the combo box, when the top level group box isn't selected first. You should be able to test this by selecting the big box first and then selecting from the combo and see if it works correctly then.

This code:
```
                                onActivated: {
                                    applyButton.primary = true
                                    controller.autostartId = modelData.airframes[index].autostartId
                                    airframeCheckBox.checked = true;
                                }
```

I think the ```airframeCheckBox.checked = true``` may be triggered the other handler to reset the id before currentIndex of the combo has changed to the new value. I don't have a board to test with right now. But I think the order may need to be flipped:
```
                                onActivated: {
                                    applyButton.primary = true
                                    airframeCheckBox.checked = true;
                                    controller.autostartId = modelData.airframes[index].autostartId
                                }
```

DonLakeFlyer, 2016-11-24:
Yup. I instrumented the code and tested with MockLink. That's the problem. You can work around this by clicking the big box first, then selecting from the combo. I'll fix it.

closed 2016-11-24 by DonLakeFlyer

clovett, 2016-11-24:
that sounds like it, thanks.