mavlink/qgroundcontrol #4171: Altitude change UI confusing/scary

julianoes, 2016-10-23:
First of all, cool. I like the ease of use of the commands at the bottom like takeoff, change in altitude, etc. I only just discovered this yesterday but it's very convenient!

Now, to my issue: I find it scary that the default for an altitude change is 100m.

Also, the slider goes from 0 to 100m. I would propose to have a slider from e.g. -50 to 50m and default to 0m. This makes a user realize that it is a relative change and not an absolute one.

dagar, 2016-10-23:
We could also limit it from going under home.

DonLakeFlyer, 2016-11-08:
Altitude change is not relative you are specifying and absolute altitude above home if I remember correctly.

NaterGator, 2016-11-08:
> Altitude change is not relative you are specifying an absolute altitude **above home** if I remember correctly.

I don't know, that sounds relative to me. Am I misunderstanding? When I think of absolute altitudes I think of something more akin to height above MSL or the WGS84 ellipsoid.

DonLakeFlyer, 2016-11-08:
Talking about two different things:
1) The altitude you specify in the slider is home relative
2) It is not a "relative" adjustment to the existing altitude. 

For example: In Julian's example of setting slider default to 0. That would cause the vehicle to go back to home altitude. It would not cause the vehicle to just stay at the same height.

julianoes, 2016-11-08:
> 1) The altitude you specify in the slider is home relative

Well, that's what I would have guessed but that's not what's happening.

dagar, 2016-11-08:
There are 2 cases that use the altitude slider.

1. `Takeoff` and enter altitude above home (MAV_CMD_NAV_TAKEOFF)
2. `Change Altitude` in flight and enter altitude relative to current (MAV_CMD_DO_REPOSITION)

julianoes, 2016-11-08:
@dagar: Right, I'm referring to 2.

DonLakeFlyer, 2016-11-08:
Ok. I get get it now. It's a bug in change altitude. Initial implementation was for ArduPilot which is **not** "relative to current" change. Then was used for PX4 as well which **is** "realtive to current."

DonLakeFlyer, 2016-11-15:
Altitude is now correctly set to home relative altitude.

closed 2016-11-15 by DonLakeFlyer

julianoes, 2016-11-15:
Thanks @DonLakeFlyer!