mavlink/qgroundcontrol #4172: Waypoint precision loss after send and load.

jgoppert, 2016-10-23:
I believe the waypoints are being sent using MISSION_ITEM instead of MISSION_ITEM_INT. This is causing a precision loss which means waypoints are shifting on me to the nearest (1e-7*6371000) ~ 0.6m increment.

[image]
[image]

```diff
diff generated.mission generated_from_vehicle.mission 
3c3,4
<     "complexItems": [],
---
>     "complexItems": [
>     ],
10,12c11,13
<                 40.429986510175752,
<                 -86.928994092729994,
<                 2.0
---
>                 40.429985046386719,
>                 -86.928993225097656,
>                 2
26,28c27,29
<                 40.429986510175709,
<                 -86.928993108184997,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928993225097656,
>                 0.30000001192092896
42,44c43,45
<                 40.429986510175709,
<                 -86.928993108184997,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928993225097656,
>                 3
58,60c59,61
<                 40.429986510175709,
<                 -86.928993108184997,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928993225097656,
>                 5
74,76c75,77
<                 40.429986510175709,
<                 -86.928993108184997,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928993225097656,
>                 7
90,92c91,93
<                 40.429986510174068,
<                 -86.928979324554973,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928977966308594,
>                 7
106,108c107,109
<                 40.429986510174068,
<                 -86.928979324554973,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928977966308594,
>                 5
122,124c123,125
<                 40.429986510174068,
<                 -86.928979324554973,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928977966308594,
>                 3
138,140c139,141
<                 40.429986510174068,
<                 -86.928979324554973,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928977966308594,
>                 0.30000001192092896
154,156c155,157
<                 40.429986510170792,
<                 -86.928965540924949,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928962707519531,
>                 0.30000001192092896
170,172c171,173
<                 40.429986510170792,
<                 -86.928965540924949,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928962707519531,
>                 3
186,188c187,189
<                 40.429986510170792,
<                 -86.928965540924949,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928962707519531,
>                 5
202,204c203,205
<                 40.429986510170792,
<                 -86.928965540924949,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928962707519531,
>                 7
218,220c219,221
<                 40.429986510165882,
<                 -86.928951757294925,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928955078125,
>                 7
234,236c235,237
<                 40.429986510165882,
<                 -86.928951757294925,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928955078125,
>                 5
250,252c251,253
<                 40.429986510165882,
<                 -86.928951757294925,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928955078125,
>                 3
266,268c267,269
<                 40.429986510165882,
<                 -86.928951757294925,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928955078125,
>                 0.30000001192092896
282,284c283,285
<                 40.429986510159338,
<                 -86.928937973664915,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928939819335938,
>                 0.30000001192092896
298,300c299,301
<                 40.429986510159338,
<                 -86.928937973664915,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928939819335938,
>                 3
314,316c315,317
<                 40.429986510159338,
<                 -86.928937973664915,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928939819335938,
>                 5
330,332c331,333
<                 40.429986510159338,
<                 -86.928937973664915,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928939819335938,
>                 7
346,348c347,349
<                 40.429986510151146,
<                 -86.928924190034891,
<                 7.0
---
>                 40.429985046386719,
>                 -86.928924560546875,
>                 7
362,364c363,365
<                 40.429986510151146,
<                 -86.928924190034891,
<                 5.0
---
>                 40.429985046386719,
>                 -86.928924560546875,
>                 5
378,380c379,381
<                 40.429986510151146,
<                 -86.928924190034891,
<                 3.0
---
>                 40.429985046386719,
>                 -86.928924560546875,
>                 3
394,396c395,397
<                 40.429986510151146,
<                 -86.928924190034891,
<                 0.3
---
>                 40.429985046386719,
>                 -86.928924560546875,
>                 0.30000001192092896
410,412c411,413
<                 40.429986510151146,
<                 -86.928924190034891,
<                 0.0
---
>                 40.429985046386719,
>                 -86.928924560546875,
>                 0
416c417
<             "param1": 2,
---
>             "param1": 0,
427,429c428,430
<             40.429986510175752,
<             -86.928994092729994,
<             0.0
---
>             40.429985046386719,
>             -86.928958892822266,
>             0

```

jgoppert, 2016-10-23:
This is what the mission looks like in 3D to give you an idea of how it is shifting the points graphically. Note that there should be equal spacing on each up/down leg, but it is off due to rounding.

[image]

dagar, 2016-10-23:
MISSION_ITEM_INT isn't currently implemented in QGC and only recently arrived in PX4.

julianoes, 2016-10-23:
> MISSION_ITEM_INT isn't currently implemented in QGC and only recently arrived in PX4.

Correct, QGC should implement it and then use by default if the autopilot supports it which can be checked using `MAV_PROTOCOL_CAPABILITY_MISSION_INT`.

dagar, 2016-10-23:
So how did you test it @julianoes? :smirk:

LorenzMeier, 2017-03-12:
@DonLakeFlyer Just stumbled across this again today - we definitely should do this since folks do tend to care about 0.6 meters. The vehicle manoeuvres more accurately than that.

Needs to implement `MAVLINK_MSG_ID_MISSION_REQUEST_INT` and then send out `MAVLINK_MSG_ID_MISSION_ITEM_INT` messages.

closed 2017-03-15 by DonLakeFlyer

jgoppert, 2017-03-15:
Nice!