mavlink/qgroundcontrol #4175: Establish agreed-on standard log download in common message set

LorenzMeier, 2016-10-23:
@magicrub this is to track our previous discussion from the MAVLink PR and make sure we have all interested projects sign off on the final implementation in QGC (which is not done yet). @dogmaphobic is working on an automatic log streamer including a log rotate (so that it can default to always on without overflowing the GCS storage).

My goal is to go with your proposal of using the same fields and business logic as you're already using, just in the common message set - if technically feasible. Right now we're tinkering on the way there with @bkueng's implementation from the PR.  I expect it will take 1-2 weeks to prototype this here and then we can take a step back and establish a plan how to go about cross-project testing and docs.

magicrub, 2016-10-23:
CC'ing people involved with original 3DR implementation: @peterbarker @bugobliterator

DonLakeFlyer, 2017-03-19:
I believe this is done now

closed 2017-03-19 by DonLakeFlyer

peterbarker, 2017-03-19:
On Sun, 19 Mar 2017, Don Gagne wrote:

> I believe this is done now

Well... let's say that there's no more work to be done before closing the 
issue :-)