mavlink/qgroundcontrol #4177: Plane skip lines i n survey

jcooper21, 2016-10-24:
I just tested out the survey functionality and it works good.  One thing we are missing is the ability for a plane to skip lines in order to reduce overshoot and increase mapping area.  What this means is if there are 5 lines in the survey mission then the plane will fly line 1, line 3, line 5 then line 2 and line 4.  Mission planner also has this function.

DonLakeFlyer, 2016-11-08:
@birchera Could this be automatically determined based on turning radius?

birchera, 2016-11-08:
Checking for grid spacing < 2 * turning radius could make sense. But from the top of my head turning radius is computed by the position controller and depends on airspeed (for px4 at least). We don't have easily access to this, at least not when planning before the flight.
I am not sure though what else there is that we could go with.

DonLakeFlyer, 2024-10-13:
Support for "Fly alternate transects" was added a while back