mavlink/qgroundcontrol #4187: Master broken: sdl2 development package not found

birchera, 2016-10-31:
I can't compile master. I get the message "Project ERROR: sdl2 development package not found"

I assume this has to do with https://github.com/mavlink/qgroundcontrol/pull/4110, but don't know how to solve it.

This is on both Ubuntu 14.04 and 16.04.

dagar, 2016-11-06:
Try installing libsdl2-dev?

birchera, 2016-11-06:
That solved it, thanks for the hint!

closed 2016-11-06 by birchera

Evanhahaha, 2019-07-04:
I also get this message and sdl2 is in installed already. This is on Mac pro

Evanhahaha, 2019-07-04:
> Try installing libsdl2-dev?

I also get this message and sdl2 is installed already. This is on Mac pro

DonLakeFlyer, 2019-07-04:
libsld2-dev has nothing to do with OSX. That is just for Ubuntu. OSX builds use an SDL framework which is part of the repo.