mavlink/qgroundcontrol #4191: Visual modifications on QGroundControl

Benbenbeni, 2016-11-01:
Hi everyone.

I'm fairly new to the QGroundControl community and I've just download the lastest version.
I would like to customize some visual aspect on the Fly panel of QGroundControl.
For example, I would like to change the plane picture in the compass (replace it by a copter or something more fun like a rafale jet) or change the colors in the artificial horizon.

I used to work on an old version of QGC (v2.5 or .7 I think). All these parameters could be accessed in a "PrimaryFlightDisplay.cc" file but it seems to have changed since then and I'm not sure 

I would like to know if someone could give me some advices about that.

Thanks in advance.

DonLakeFlyer, 2016-11-08:
Please, bugs only here. Discussion: http://discuss.px4.io or https://gitter.im/mavlink/qgroundcontrol. 

UI is written in Qml now. Look at FlightDisplayViewWidgets.qml for how the instruments get added.

closed 2016-11-08 by DonLakeFlyer

Benbenbeni, 2016-11-09:
That's what I thought but i wasn't sure where to ask.

Thanks for the tip.