mavlink/qgroundcontrol #4195: gstreaming working but not showing up in qgroundcontrol

RickBlacker, 2016-11-06:
Hi all, i'm following these instructions..
https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md

I have a gstreamer wifi stream working between two computers.  
First I start gstreamer on my Ubuntu machine running QGC.
Next I start the stream on my other linux box.  
On my machine running QGC, a new window pops up displaying my stream. However, the stream is not showing up in QGC, rather a separate window.

I did NOT compile QTC, rather Iam using the QGC Appimage.

**This is what is running on the machine with QGC**
gst-launch-1.0 udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! autovideosink fps-update-interval=1000 sync=false

**This is what is running on my stream server (What will be the drone)**
gst-launch-1.0 v4l2src device=/dev/video13 do-timestamp=true ! video/x-raw, format=YUY2, width=640, height=480, framerate=30/1 ! autovideoconvert ! vaapih264enc ! rtph264pay ! udpsink host=192.168.1.2 port=5600

Do you guys have any suggestions?
Thanks!!

jaxxzer, 2016-11-08:
QGC will connect to the stream automatically. You don't need to run gstreamer separately on the QGC machine. Your script is probably binding the port before QGC can.

dogmaphobic, 2016-11-08:
This was duplicated on Gitter and resolved there.

closed 2016-11-08 by dogmaphobic