mavlink/qgroundcontrol #4196: QGC won't connect to Windows 10 COM port with 3DR SIK radio running 115200 baud even after setting baud rate to 115200

BigOkie, 2016-11-06:
Running QGC 3.0.2 on Windows 10 with Anniversary Update.  QGC runs/connects fine if just connecting using the USB port.

Attaching consolelog with pertinent messages.  Please inform if I need more messages in the log and I will provide them.  The gist is that on initial blush it asks about baud rate mismatch (I guess it starts at 57600 for the radio); when I change the baud rate to match, then it says it cannot create the port and that access is denied.

[qgclog.txt](https://github.com/mavlink/qgroundcontrol/files/573981/qgclog.txt)

DonLakeFlyer, 2016-11-08:
What firmware type/version are you using?

BigOkie, 2016-11-08:
Sorry.

Arducopter 3.4.1 stable.

DonLakeFlyer, 2016-11-08:
Have you turned on Mavlink 2.0 support on the vehicle you are connecting to? Stable 3.0.2 doesn't yet support Mav 2, but Daily builds do.

BigOkie, 2016-11-08:
No they were set to Mavlink 1 via Serial0 and Serial1 protocol.

DonLakeFlyer, 2016-11-09:
FYI: The correct baud rate for radio is 57600.

I just tried to repro on Windows and OSX as well. 3.0.2 + 3.4.1 worked fine on both. Still seems to point to a firmware setup issue as opposed to QGC. But not sure what. Could you try installing APM Planner 2 and see if you can connect with that?

BigOkie, 2016-11-09:
At work currently but will look when I get home.

Is there a reason that the 'correct' baud rate for the radio is 57600, when instructions are also given to change that to 115200?  I'll change it back if needed, but it always gives me issues when I do after I changed it to 115200.

DonLakeFlyer, 2016-11-09:
> Is there a reason that the 'correct' baud rate for the radio is 57600, when instructions are also given to change that to 115200?

Are you talking about the message QGC pops up, or something else that is telling you to switch to 115200? If the QGC message, in this case QGC can't determine the difference between a baud rate mismatch and a mavlink protocol screwup. So it says what the possible problems are. it may be wrong.

BigOkie, 2016-11-09:
It doesn't tell me to switch to it; the first message I get regards a baud rate mismatch or protocol version mismatch.  I'll try and change the radio to 57600 when I get home and report what happens after that.

BigOkie, 2016-11-10:
Switching the radios to 57600 solved the problem.  I do wish it would work with 115200 though.  Both APM Planner and Mission Planner do, as well as QGC on Android.

DonLakeFlyer, 2016-11-10:
You should be able to change to 115200 if you want. But you will need to create a manual connection for that. Auto-connect always connects at 57600. You will also need to turn off auto-connect for SiK Radio (Settings/General). Then create the manual connection as connect with that. Let me know if that works for you.

BigOkie, 2016-11-10:
That doesn't work in Windows however.  I've tried it numerous times (changing the baud rate when I create a manual connection).  I get the same results.  It works just fine creating/modifying the connection to 115200 on Android thoiugh.

DonLakeFlyer, 2016-11-10:
Did you turn off auto-connect for Sik Radios? If so, can you attach a log for the manual connect case with LinkManagerLog turned on?

BigOkie, 2016-11-10:
Actually, I turned off the autoconnect just now, created a manual connection and that appears to have solved it.  Not sure why you can't change it when it's autoconnecting but this is a workaround I can use.

Thanks for your patience on this.

DonLakeFlyer, 2016-11-10:
When auto-connect is on, it will connect to the port automatically even if the baud rate is wrong. This will prevent your manual connection from connecting a second time since the port is already open. Hence the "access is denied" message.So QGC is doing the right thing in this case even though it's not super easy to understand why you get the "access denied" error.

DonLakeFlyer, 2016-11-10:
I'm chaging this issue to give a better error if you manually connect to the same port which is set up for auto-connect. I think it's possible to detect that and give a better error.

BigOkie, 2016-11-10:
Once again, thanks for your patience.  I like both Mission Planner and QGC, but I feel like QGC on Windows is a lot easier to use, especially creating missions.

DonLakeFlyer, 2016-11-10:
No problem at all. It's really your patience as well that helped to get to the bottom of this. And helped to determine a bad usability problem that we'll fix.

Keep the problems and/or comments coming.