mavlink/qgroundcontrol #4197: Videostreaming requirements needs to be refined

AndKe, 2016-11-07:
https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md
says:
 sudo apt-get install gstreamer1.0*

That's borderline crazy, it would mean:

gstreamer1.0-alsa                      gstreamer1.0-plugins-bad               gstreamer1.0-plugins-ugly
gstreamer1.0-clutter-3.0               gstreamer1.0-plugins-bad-dbg           gstreamer1.0-plugins-ugly-amr
gstreamer1.0-crystalhd                 gstreamer1.0-plugins-bad-doc           gstreamer1.0-plugins-ugly-dbg
gstreamer1.0-doc                       gstreamer1.0-plugins-bad-faad          gstreamer1.0-plugins-ugly-doc
gstreamer1.0-dvswitch                  gstreamer1.0-plugins-bad-videoparsers  gstreamer1.0-pocketsphinx
gstreamer1.0-espeak                    gstreamer1.0-plugins-base              gstreamer1.0-pulseaudio
gstreamer1.0-fluendo-mp3               gstreamer1.0-plugins-base-apps         gstreamer1.0-rtsp
gstreamer1.0-hybris                    gstreamer1.0-plugins-base-dbg          gstreamer1.0-rtsp-dbg
gstreamer1.0-libav                     gstreamer1.0-plugins-base-doc          gstreamer1.0-tools
gstreamer1.0-libav-dbg                 gstreamer1.0-plugins-good              gstreamer1.0-vaapi
gstreamer1.0-nice                      gstreamer1.0-plugins-good-dbg          gstreamer1.0-vaapi-doc
gstreamer1.0-packagekit                gstreamer1.0-plugins-good-doc          gstreamer1.0-x

- Please narrow down to what is actually needed.

dagar, 2016-11-07:
I had it stripped down at one point, but things were apparently missing. https://github.com/mavlink/qgroundcontrol/commit/815b465d5bc3a3d971c5069a16427f649998e9d9#diff-b230c70b3e357996e3bbe30ca1260527

Would you like to help us figure out the actual minimum required gstreamer packages?

AndKe, 2016-11-08:
Yes, I can help - but I need to know what to look for.

I know that a month ago, or two, when compiling on my laptop, then QGC displayed my laptop's camera feed right away (found USB device)
Now when I compile on the laptop, I get a black window, - and it does not display a stream from a Solo (tested only that)
So: what is the expected behaviour ?   - and can I make a stream locally for testing, maybe using VLC ?  (command that is expected to work please.)

AndKe, 2016-11-08:
Ok, I've narrowed it down a lot, - now I only need a way to test, (to generate typical stream)

closed 2026-10-04 by github-actions[bot]