mavlink/qgroundcontrol #4200: qgroundcontrol's app problem

waterflyfish, 2016-11-10:
When I used the android mobile phone to communicate with  the px4 through  serial port ,about twenty minutes later ,the qgroundcontrol will crash.Is there anybody meet this problem,and how to solve it?

dagar, 2016-11-10:
Is there any error message? Are you connected to a vehicle the entire time?

waterflyfish, 2016-11-11:
Yes，I connected to the vehicle and use the Release V3.1.About twenty minutes later ,it will crash without any operation

jilinggui, 2016-11-15:
what time about Version 3.1 release  ？I look forward to it.

DonLakeFlyer, 2016-11-18:
@uavcam Are you having this problem as well? 

Are you connected to the vehicle the whole time? Does it crash when you are doing nothing with the tablet. Or are you clicking on something and then the crash.

pynki, 2016-11-20:
@DonLakeFlyer see the gitter private message...

just wait 15 minutes and then it crashes. thats what i can say now. not sure about if the connection - i will find out for you.

BrainyBeeLy, 2016-11-29:
I think the bug is caused by the serial port. When using the udp connection without this problem.

DonLakeFlyer, 2016-11-29:
Whomever is seeing this problem, please make sure you are running a daily build and then report back if there is still a problem.

waterflyfish, 2016-11-30:
It would report the error "eglSwapBuffers failed" when crashed.

DonLakeFlyer, 2016-11-30:
Are you using a daily build?

waterflyfish, 2016-11-30:
yes

DonLakeFlyer, 2016-12-01:
I think I have a repro for this.

DonLakeFlyer, 2016-12-04:
Should take a few hours for the fix to work it's way out into a daily build. Can you retest tomorrow with a new daily build?

closed 2016-12-04 by DonLakeFlyer

waterflyfish, 2016-12-05:
ok,I'll give you a reply as soon as possible.