mavlink/qgroundcontrol #4211: Survey, ignoring first polygon points

AndKe, 2016-11-14:
Please see video https://youtu.be/lOfMzZxgUlc

AndKe, 2016-11-14:
This may not be typical use to hit "survey" as the first thing, but it will be when you release 3.1 and users "discover" this great feature.

This does not happend if first creating a plain WP (deleting it) then creating a survey.  - may be linked to the existence of something before.

The auto-instertion of takeoff should be consistent regardless of type of mission item.

DonLakeFlyer, 2016-11-15:
Are you able to repro this on a desktop OS? Seems ok for me on OSX. Charging up my tablet to try.

Also how many points does it lose? It's really hard to see without a mouse.

AndKe, 2016-11-15:
Only on Android (Android 5.1.1 and 6.0.1 tested, on tablet and phone)
- not on Linux.

nanthony21, 2016-11-16:
The MouseArea QML type (http://doc.qt.io/qt-5/qml-qtquick-mousearea.html) that handles polygon drawing really isn't intended for touch devices. It would be more work but it may be nice to reimplement this using the Multipointtoucharea qml type. I'm not sure but I think this will also work with a mouse.

I'd be willing to work on this, I'm just not sure when I'll get around to it.

DonLakeFlyer, 2016-11-17:
> The MouseArea QML type (http://doc.qt.io/qt-5/qml-qtquick-mousearea.html) that handles polygon drawing really isn't intended for touch devices.

Where does it say that? I don't think that is the case. If all you want is a single touch point, MouseArea should work fine. It is used extensively throughout the codebase without problem. You should only need to use MultiPoint if you really want multi-point touch.

DonLakeFlyer, 2016-11-18:
@AndKe Can you test tomorrow. Should be enough time for a new daily to roll through.

AndKe, 2016-11-19:
Works perfectly , thanks.

AndKe, 2016-11-20:
@DonLakeFlyer  
Please remember that (, regardless of OS , after application load) 
-adding a waypoint consumes first click and adds takeoff. 
-adding a survey - does not add takeoff.

Is this part of user experience intentional ? - I'd suggest that:
Adding WP or survey, do instantly add Takeoff.

DonLakeFlyer, 2016-11-20:
Ah, ok I missed that. I'll fix.

AndKe, 2016-11-20:
Related to that of adding autotakeoff: 

1 User open QGC, locates the area where the planning is to be done (zooms in).
2 User adds WP - and home+takeoff is still at the old /previous place QGC have been used.
- a fresh to QGC user would now see a line, zoom out , go and drag Home to there he actually started planning. (not even seeing that Takeoff item have been created out of sight)  mission statistics are already displaying time/distance based on old home position.

- an experienced user, would open mission item, "planned home position"  and click "move home to map center"  

I do not see an advantage of keeping old home position when planning new flight for nigher user..

This would be much more intuitive, and present the user with an easier to understand result:
1 User open QGC, locates the area where the planning is to be done (zooms in).
2 User create WP(or survey)  - home/takeoff spawns in the middle of the visible map, (user sees it instantly, and are aware of both - moreover, mission statistics are instantly "correct" , or at least not off by very much)

So. please consider calling "move home to map center" when creating first WP/Surwey item.

DonLakeFlyer, 2016-11-20:
@AndKe Can you move your home position comments here: #3336? Then we can continue there. Home thing is kind of a mess. like your idea. I'll re-open for 3.1.

closed 2016-11-23 by DonLakeFlyer