mavlink/qgroundcontrol #4214: Windows 10 - QGC can't load stored auto mission waypoints from FC; also some instances saving waypoints doesn't work

BigOkie, 2016-11-14:
As the title says.  If I connect and try to read stored waypoints, I'm never able to.  Writing waypoints created in QGC is hit and miss.  I usually get the following errors when trying to read.

[image]
[image]

DonLakeFlyer, 2016-11-15:
Could you turn on MissionControllerLog and MissionManagerLog logging (https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html) and get me log output for both cases? Seems to be working ok for me.

DonLakeFlyer, 2016-11-15:
Also detailed order of repro steps. Start QGC first, then plug in board. Plug in board, then start QGC that sort of thing.

BigOkie, 2016-11-15:
Will do this when I get home this evening.  Thanks.

BigOkie, 2016-11-15:
1. Open QGC (not configured to auto-connect due to my radios using 115200 baud)
2. Power the FC (in this case a 3DR Pixhawk1)
3. In QGC connect to the radio.
4. Switch to the flight plan view in QGC
5. Try to load waypoints (this sometimes works and sometimes doesn't; in this case it did)
6. If it does work, clear the waypoints (this works if the waypoints load)
7. Create a new mission with some waypoints and try and send (in this instance, it sent them ok and I verified it in MP)
8. Returned to QGC after verifying in MP that the waypoints were indeed there and now QGC will not load them.

I have attached the log file generated with the options you asked for set.

[qgcmission.txt](https://github.com/mavlink/qgroundcontrol/files/593360/qgcmission.txt)

BigOkie, 2016-11-16:
Also noticing it does the same on Android.  I have a Nexus 10 running Marshmallow.

DonLakeFlyer, 2016-11-17:
Clarifying what you are doing in step 8, looks like:
* Boot QGC
* Move to Plan view
* Then maybe you click on Sync?
Is that correct?

DonLakeFlyer, 2016-11-17:
Also, can you try to repro the problem when connect via USB instead of radio?

DonLakeFlyer, 2016-11-17:
Current thinking is that this is radio packet loss causing problems with QGC mission protocol not hardended correctly to protocol failures and possible retry logic.

BigOkie, 2016-11-17:
It appears to work fine with USB...only with radio does it appear to have problems.

And yes, my order is:

Start QGC
Connect to FC (either method_)
Move to plan view
In USB, moving to plan view seems to automatically download the waypoints if they are there.  Not so much when using the radio and in some instances it will throw the error.

DonLakeFlyer, 2016-11-18:
@BigOkie New daily build with fixes for this should be available in a couple hours.

DonLakeFlyer, 2016-11-23:
@BigOkie Any news on daily builds fixing this problem for you?

BigOkie, 2016-11-23:
Some extent I think, yes.  I have a logfile I can attach however creating and reading missions seems to work better.  On occasion, I will get an error like this:

[E] at qrc:qml/MissionEditor.qml:735 - "qrc:qml/MissionEditor.qml:735:21: QML ListView: Cannot set a negative cache buffer"

Other times I see this when trying to upload a mission:

[D] at :0 - "Sending error 5 "Mission write failed, vehicle failed to send final ack."

But it still appears to send the mission.

I'm attaching the log from me testing the daily build I installed just today.
[console.txt](https://github.com/mavlink/qgroundcontrol/files/607983/console.txt)

DonLakeFlyer, 2016-11-24:
> [E] at qrc:qml/MissionEditor.qml:735 - "qrc:qml/MissionEditor.qml:735:21: QML ListView: Cannot set a negative cache buffer"

This can be ignored.

> [D] at :0 - "Sending error 5 "Mission write failed, vehicle failed to send final ack."
 
Given the way the write protocol (pull protocol) works there doesn't seem to be a great way to fully harden it against packet loss. I'll do some more poking around on that. Read side should be much better now though.

BigOkie, 2016-11-24:
I'll continue to poke around with the dailies and create missions in different planners (Tower, MP, etc) and see how it works out.  Thanks for the help again.

DonLakeFlyer, 2016-12-30:
There are more complicated possibilities for write side. Moving to 3.2

closed 2026-10-04 by github-actions[bot]