mavlink/qgroundcontrol #4222: Android: UDP video does not show on one device.

AndKe, 2016-11-18:
I have tested UDP video streaming from 3DR Solo to two devices using QGC for android beta from google play:

phone (Android 6.0.1):  (fail) 
-start QGC, no video - quit
-start Solex - video ok, quit
-start QGC, no video - quit

Same procedure on tablet (Android 5.1.1)  is pure success

Please tell how I can help you narrow it down.
Logcat did not help much, but revels maybe something else. 
[image]

dogmaphobic, 2016-11-18:
There isn't much to go on. The warnings above are not related to video streaming. I have not seen or heard of any video issues until this one. Have you tried using a test video source such as
```
gst-launch-1.0 videotestsrc pattern=ball ! x264enc ! rtph264pay ! udpsink host=xxx.xxx.xxx.xxx port=5600
```
Replacing the xxx.xxx.xxx.xxx with your phone's IP address.

Also, try turning the phone cellular modem off to see if it makes any difference (use only WiFi).

AndKe, 2016-11-18:
Thank you - I regret to say, this only adds to the mystery.

The command to generate a stream, works perfectly for both devices.
I have tried to make the phone to work for a while now, rebooting it , force-stopping any the other GCS software that I suspected might have somehow be hooking up to the stream first.

Also, forgot to mention; both phone and tablet have SIM and network from the same provider, so if there were a routing related problem - none should have worked, but this is a 10.* net , so it's hard to imagine - also - Solo, Solex, and Tower display the stream properly on the phone any time...

Any other suggestions ?

AndKe, 2016-11-18:
@dogmaphobic - I am impressed (with you), - just to left nothing untested, I disabled the cellular network
- started QGC - works fine !   - quic QGC
- enabled cellular
- started QGC - does not work !   - quic QGC

I tried this two times.  And YES, QGC is somehow the only application that fails to connect due to the cellular internet access being available.

10.0.0.0/8  is private, so I cannot imagine it being routed wrongly, moreover, if it were, then the other GCS applications and telemetry stream should have failed too, but this is only video, on only this device.

Please try to come up with some idea about what makes this application behave differently..

dogmaphobic, 2016-11-18:
No, it just rung a bell from some other issue related to being able to get access to maps while connected through wifi to a device with no Internet (or the other way around, now I can't remember). There was a change in Android 6 that caused this. Tower and the others are true Java/Android apps and use the system API to access the network. QGC is c++, running below it all on top of the Linux kernel. It uses a wrapper to work on Android. There ought to be some (Java) API to tell Android to use the WiFi link even if it has no access to the Internet (this is what Tower and such must be doing).

I don't have anything Android with cellular network to test this though...

DonLakeFlyer, 2016-11-18:
FYI: https://github.com/mavlink/qgroundcontrol/issues/3128

AndKe, 2016-11-27:
Please look at the following code; 

https://github.com/DroidPlanner/Tower/blob/671730426d00cdf8b1063c879266ea80bd798c3c/Android/src/org/droidplanner/android/AppService.java#L126

Tower checks for a "sololink" - then , does some magic depending on Android version..

AndKe, 2016-11-27:
Also line 146, 158, 163 - all are based on some Android version.

DonLakeFlyer, 2016-11-29:
I don't any reference to sololink in that file. That code just looks like more handling on WiFi connection without internet.

AndKe, 2016-11-29:
line 49 and 89 contain "if (NetworkUtils.isOnSoloNetwork(context)) {"
it leads to 
Android/src/org/droidplanner/android/utils/NetworkUtils.java:37:    public static boolean isOnSoloNetwork(Context context)

Anyway, I am not sure the problem is related to *sololink* itself, but the fact that "something else" needs to be done when on certain Android versions (so I were looking for something that other software do, but QGC does not. )

AndKe, 2016-11-29:
Please observe the different versions it's handling:
```
Android/src/org/droidplanner/android/AppService.java:127:        if (Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP)
Android/src/org/droidplanner/android/AppService.java:146:                if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
Android/src/org/droidplanner/android/AppService.java:158:        if (Build.VERSION.SDK_INT < Build.VERSION_CODES.LOLLIPOP)
Android/src/org/droidplanner/android/AppService.java:163:        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
```

DonLakeFlyer, 2016-11-29:
Thanks I see it now. I still think that's all either WiFi connection without internet, or they may have found a way to support WiFi to Solo plus cell network for internet. It fundamentally doens't make sends that you can get telemetry but not video since that is all over the same WiFi connection.

When you run into this problem can you check the debug console: Settings/Console. There should be a line in there that says "Found Solo". The indicates it is setting up the special solo video handshake.

AndKe, 2016-11-30:
As for being a routing error - no , it makes no sense.
But what you wrote, was very helpful.
Android app fails to detect it.
[image]
QGC on Linux does: 
[image]

AndKe, 2016-11-30:
may it be that the APMFirmwarePlugin::_handleIncomingStatusText fail due to the errors ?
in case, here is the rest of the console output:
[image]

DonLakeFlyer, 2016-11-30:
Can you install this build: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/OneOffBuilds/QtApp-debug.apk

This will output to the console all the status text input. Then take a look for the Solo version message and post it here. It should start with this "APM:Copter solo-".

AndKe, 2016-11-30:
I did already check what Solo outputs; 
I think it is "APM:Copter solo-1.3.3"  or thereabout.
in any case, current master on desktop detects it just fine using:
```
./src/FirmwarePlugin/APM/APMFirmwarePlugin.cc:27:static const QRegExp APM_SOLO_REXP("^(APM:Copter solo-)");
```
So the string itself is not the problem.  (If you still believe it may be it, I'll test tomorrow at work - I don't have a solo at home now)

AndKe, 2016-11-30:
BTW, found a tlog - the exact statustext is "APM:Copter solo-1.3.1 (7e9206cc)"

DonLakeFlyer, 2016-12-01:
Yeah I know what the string is. I want to know if QGC on Android is seeing this status message or not. Hence the custom build. Seems like it is not getting that message.

AndKe, 2016-12-01:
This is strange. 
essential messages are missing, but other statustext gets thru:
[image]

[image]

[image]

FYI: "communication lost" is a result of the prearm warning always scrolling to the end, (and speaking) - that made me shut down Solo - even that message is fine.

BTW: "Save App Log" - displays a rather unusual file dialog.

DonLakeFlyer, 2016-12-02:
Ok, this makes sense now. A single status text message just isn't a good way to detect solo. Over a not so good WiFi link it could be lost and never seen. Need to find something else as well to identify solor and trigger solo video start.

AndKe, 2016-12-03:
Well - while I agree that UDP packet loss is acceptable, it's strange that QGC on phone seems to be the only device/GCS that fail to recieve that message. 
Besides the fact that a Solo could be recognized by ssh to 10.1.1.1 (controller /(router)) or the solo itself (10.1.1.10)  - there is one funny configuration that you could use to detect solo: 

SERIAL1_BAUD,921  (high speed to its companion computer)

and all modes are Loiter.

FLTMODE1,5
FLTMODE2,5
FLTMODE3,5
FLTMODE4,5
FLTMODE5,5
FLTMODE6,5

while it's not completely reserved to Solos  - it is at least a rather uncommon config.

Still, I am near to the controller(router) (which is close to solo) when connecting, it seems to lose those texts only on the Android phone, only with QGC , which is very strange - I can do some more testing in a row - to see if it at least once succeeds... (which could confirm the Wifi packet loss theory.)  I must say I doubt it.

AndKe, 2016-12-05:
I did more testing.  (with the daily android build, on the same 6.0.1 device.
Yes, it works - must have been due to the lost statustext - I cannot even reproduce the problem anymore.

AndKe, 2016-12-13:
One of the Tower devs pointed me to https://developer.android.com/about/versions/android-5.0.html , look for "Multiple network connections".

dogmaphobic, 2016-12-28:
This exceeds my knowledge of Android/Java. I found the documentation explaining the issue here:

https://developer.android.com/reference/android/net/wifi/WifiManager.html#enableNetwork(int,%20boolean)

The paragraph is quite self-explanatory:

*Note: If an application's target SDK version is LOLLIPOP or newer, network communication may not use Wi-Fi even if Wi-Fi is connected; traffic may instead be sent through another network, such as cellular data, Bluetooth tethering, or Ethernet. **For example, traffic will never use a Wi-Fi network that does not provide Internet access (e.g. a wireless printer), if another network that does offer Internet access (e.g. cellular data) is available.** Applications that need to ensure that their network traffic uses Wi-Fi should use APIs such as bindSocket(java.net.Socket), openConnection(java.net.URL), or bindProcessToNetwork(Network) to do so.*

Out of the suggested solutions, it seems only `bindProcessToNetwork(Network)` may be applicable, assuming it will affect how the Qt (c++) network interface interacts with it. I will see in what kind of trouble I can get myself into...

dogmaphobic, 2016-12-28:
Also note that this only relates to the ability to use WiFi for connecting to the vehicle while using Cellular for Internet access (Maps, etc.) Issue #4355 is not directly related. That has to do with the overall ability to have the device connecting at all to a WiFi AP with no Internet.

dogmaphobic, 2016-12-28:
> Note: If an application's target SDK version is LOLLIPOP or newer...

QGC currently targets LOLLIPOP (API Level 21). I wonder if setting it back to KitKat (API Level 19) would solve this? I just built a version of QGC set to API level 19 and it seems it all works just fine. I don't have an Android device with cellular data though so I cannot test to see if it handles both WiFi and Cellular at the same time.

AndKe, 2016-12-29:
The Solo my company has (only one for development right now) is in pieces, being adapted to the payload the  re supposed to carry if used professionally. So I am unable to test right away, - but will test sooner or later.
As you see from my post 24 days ago, I could no longer reproduce the problem in such a way that it appeared to have anything to do with cellular connection anyway.

dogmaphobic, 2020-02-11:
Ancient history

closed 2020-02-11 by dogmaphobic