mavlink/qgroundcontrol #4228: Connect to UDP server as client

Sulter, 2016-11-20:
In most cases when connecting to a drone that is on a mobile network (i.e. like in the case of this years Outback Challenge), you would usually want to use UDP. You would then either use some 'cloud service' or have a static IP on the drone, which you would connect your ground station to.
It would be great if QGC supported use case like these by allowing connection to a UDP server.

dogmaphobic, 2016-11-20:
It's not intuitive as it should be but it's there. QGC works on the assumption it will receive one or many connections from one or multiple vehicles. So it starts as a "server" listening for connections on a specific port. You can however, manually add target hosts. You can enter either a fully qualified host name or an IP address and QGC will "call" them.

[image]

Sulter, 2016-11-20:
This doesn't seem to work for my setup.
I've tried both with Mavlink 1 as well as Mavlink 2.
I'm connecting as UDP client just fine using Mission Planner and APM Planner in those cases (except Mavlink 2 which doesn't work with APM Planner).
I'm using the current daily build on linux and android (tried both with no luck).

DonLakeFlyer, 2016-11-21:
Just to fill in the details can you provide a screenshot of you comm link settings you are using for UDP? Also can you provide details on how you are connecting via APM Planner?

Gus, if the listening port is not changed (default). Do you need to  turn off udp auto-connect in order to get a manual UDP port to work? Wondering if they will conflict. This is the case for radios on a serial port. You need to turn on radio auto-connect to get your manual serial port to work.

dogmaphobic, 2016-11-21:
In theory, you should be able to "Edit" the existing "Default UDP" connection and add a target host. When I tried that, I ended up with multiple "Default UDP" links.

In order to edit a link, you need to disconnect it first. As soon as you disconnect  one of those "automatic" links, a new one is created underneath you. You should be able to edit the link without having a new one being created under you.

If you do create a separate link using only outgoing connections, that link should use a different port than the default one unless you disable the automatic link.

With all that said, I need to find something to test this with. I don't have anything acting as a vehicle that expects to be called in as opposed to be calling out.

DonLakeFlyer, 2016-11-21:
@Sulter I'd try this:
* Turn off UDP auto-connect, Setup/General page
* Create a new UDP link setup like Gus describes with the target host
* Manually connect to that

That removes all these automatic link problem from the equation.

DonLakeFlyer, 2016-11-23:
@Sulter Daily builds are updated with the heartbeat fix. Can you try this again?

DonLakeFlyer, 2016-11-30:
@Sulter Do you have time to verify this?

DonLakeFlyer, 2016-12-03:
Closing assuming it's fixed.

closed 2016-12-03 by DonLakeFlyer

Tutorgaming, 2017-03-02:
@dogmaphobic  I have some question from this context.

> It's not intuitive as it should be but it's there. QGC works on the assumption it will receive one or many connections from one or multiple vehicles.
 _**So it starts as a "server" listening for connections on a specific port.**_
You can however, manually add target hosts. You can enter either a fully qualified host name or an IP address and QGC will "call" them.

1.If I want to control multiple vehicles,
Can I specify just only the port on the server that I want to open for robots?

Like creating a UDP Client (at the robots with server_address = 192.168.1.1:33333) 
and use QGC as the server.

Could this scenario be possible ?