mavlink/qgroundcontrol #4229: Vehicle health no available once gone.

AndKe, 2016-11-21:
Reproduced by tlog playback:
When QGC loads, I can see 
[image]
Once clicked - it's gone:
[image]

it seems like there is no way to get information about possible bad health from MAV_SYS_STATUS_SENSOR anymore..

DonLakeFlyer, 2016-11-21:
What do you mean by once clicked? You should be able to swipe the center panel to the right to get it back. That isn't working?

I do see another bug in here which shows in the first image. Vehicle health is the second page, but the page indicator is showing the first page selected. That maybe part of the swipe bug as well if that's what is going wrong.

AndKe, 2016-11-21:
Wow ! I misunderstood the UI.
Yes, I should have guessed the three dots below were "pages" - but given being on desktop os  , I never considered to use mouse drag on those - I did, however try to click them.  (clicking anywhere within loads "previous" page, usually the first tab (voltage , GS++)   - where I thought I was "stuck"  :)
Also - I did try to click those dots :)

Swiping is just less used on computers, and can be pretty hard on small area on laptop use.

Please make clicking on the "pages" not always load the previous page, but maybe rather "next page"  - and rollover to first from last page.   That would be very intuitive.

DonLakeFlyer, 2016-11-22:
Yeah I can change that.

DonLakeFlyer, 2016-11-24:
> Please make clicking on the "pages" not always load the previous page, but maybe rather "next page" - and rollover to first from last page.

Done.

closed 2016-11-24 by DonLakeFlyer