mavlink/qgroundcontrol #4230: Look at GeoTag support for plugin-izing

DonLakeFlyer, 2016-11-21:
Currently only supports PX4 logs.

DonLakeFlyer, 2016-12-27:
Good enough for now

LorenzMeier, 2017-05-01:
@DonLakeFlyer The more generic approach is this one:

* Autopilot sends http://mavlink.org/messages/common#CAMERA_IMAGE_CAPTURED
* QGC stores all that data in a QHashMap with the sequence as key
* It re-requests any dropped image frames using http://mavlink.org/messages/common#MAV_CMD_REQUEST_CAMERA_IMAGE_CAPTURE
* It stores all that data in the flight meta info
* It uses that in the geo-tagging, which now works with any MAVLink enabled autopilot (and hence doesn't need to be a plugin)

It also means no more log downloading - you can geo-tag right away after landing. Just point QGC to the mounted SD card and it can do the magic.

mhkabir, 2017-05-01:
We should support a minimum of these EXIF tags : https://support.pix4d.com/hc/en-us/articles/205732299#gsc.tab=0

dagar, 2017-05-01:
These might be obvious, but when I see trying this kind of thing with cheap cameras and triggers I see a few common problems.
 - camera time is often wrong and at best only 1s resolution. File modification time is often lost after the photos have been manually moved around.
 - ignoring camera time and trusting the camera triggers from the autopilot works well, but only if you can get them to align and didn't drop any shots (IR trigger + bright day). There are often test shots before the log started.
 - there's a delay between the autopilot camera trigger message and the actual photo. It's variable, and dependent on all the pieces (trigger, camera, camera settings), but people will want to add this nudge factor.

For tagging I think you need to try and align the list of camera tags with the images, while only trusting the time delta between photo timestamps. It needs to be clear to the user what's happening, and if they need to manually select the first photo.

LorenzMeier, 2017-05-03:
@mhkabir Triggers get received here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L632-L643

So that's where the re-request should happen. And the geotagging plugin should take this trigger list. @DonLakeFlyer that's where your feedback is required.

DonLakeFlyer, 2017-05-04:
The fundamental problem with this is that you can't rely on telemetry to tag against since it can be lossy. It is also not an uncommon Survey use case for the vehicle to go out of telemetry range due to geography. You want a non-lossy data stream to tag from which is why vehicle binary logs are used in almost all cases.

birchera, 2017-05-05:
We already do something similar and I am happy to share our experience: All tags are stored on the aircraft during the flight. Once the aircraft lands and disarms we transfer all the tags. At that point it is usually well in range of telemetry connection.

Obviously the improved version would be to already transfer as much as possible during flight and only fetch the remaining tags after landing.

mhkabir, 2017-05-05:
How are you storing them onboard?

birchera, 2017-05-05:
As mentioned [here](https://github.com/PX4/Firmware/pull/7164/files#r115006976) we use the datamanager to store them.

DonLakeFlyer, 2017-05-09:
With ulog format geotagging going I don't see this as needed for 3.2?

LorenzMeier, 2017-05-09:
Yep, not desperately needed.

DonLakeFlyer, 2017-05-09:
I don't see how the concept of re-requesting missed CAMERA_IMAGE_TRIGGERs from the ground station side of things is a good idea. This means that QGC needs intimate knowledge of a survey which it is currently flying. Once a survey is sent to the vehicle side there is no concept of where the camera triggers are from the Fly View side of things. The Fly View is dumb to all of that. It doesn't even know it is flying a Survey.

This would require a ton of work on the QGC side of things to make it work. What exactly is the problem you are trying to solve here?

LorenzMeier, 2017-05-09:
Its not really a lot of work. What we definitely need to solve on the fundamental level is the download of flight meta data in a reliable fashion.

DonLakeFlyer, 2017-05-09:
> Its not really a lot of work.

Uh, ok. If you say so!

Why not keep camera triggers in a separate log on the vehicle. Then it's quick to download from the vehicle when the flight is done. And you don't have to do all sorts of crazy stuff to detect missing triggers.

DonLakeFlyer, 2017-05-09:
I just realized there was an image index in CAMERA_IMAGE_TRIGGER. Duh!