mavlink/qgroundcontrol #4231: Flight monitoring "Current /consumed" swapped/incorrect for plane

AndKe, 2016-11-21:
Playing back an arduplane log: (APM: ArduPlane V3.3.0 (6be0932d))

[image]

"Current", displays really "consumed capacity"    - while "consumed"  remain "-1"

This works better for ArduCopter
Current does display actual current, but in mA (it would be more readable to disaply as A, maybe with one decimal.)
"Consumed" is not computed by Arducopter, - so I don't expect it to show.

DonLakeFlyer, 2016-12-04:
These are coming straight from the mavlink messages: 
* MAVLINK_SYS_STATUS.current_battery 
* MAVLINK_BATTERY_STATUS.current_consumed

If the values are incorrect then it seems like ArduPilot is sending the wrong values. Do these show up anywhere in Mission Planner with correct values?

AndKe, 2016-12-04:
I don't know for mission planner, (don't run windows)
- but the lack of MAVLINK_BATTERY_STATUS.current_consumed  is copter is well known, unless it's recently implemented, it's just a feature that is missing.  I know, as I looked for it for mav Mav>HoTT adapter code, and ended up calculating in the adapter for copter.

I can do a fresh test on current AP and AC and let you know.

AndKe, 2016-12-04:
AP 3.7.1:
MAVLINK_SYS_STATUS.current_battery 1037  = 10.37A
It also has .battery_remaining , (percentage %) - Which I don't think is very useful for very much other than display. (does not really tell the actual capacity spent during flight w/o calculation)

MAVLINK_BATTERY_STATUS.current_consumed  does not exists on AP 3.7.1
- at least I can't see it in mavlink inspector..

AndKe, 2016-12-04:
AC 3.4.2
same messages in SYS_STATUS
I do not see BATTERY_STATUS on nigher system, using AP2 or QGC, I don't see it in mavlink inspector.

AndKe, 2016-12-04:
Sorry for all messages. 
Finished testing, actually,  QGC displays current correctly for both recent AP and AC builds - I must have played back some old logs when testing before.
"Consumed" (mAh) is not - AFAIK , transferred from AP and AC.  (maybe only on PX4 stack)

DonLakeFlyer, 2016-12-04:
@AndKe Thanks a ton for spending the time to verify. Unfortunately QGC at this point doesn't differentiate between a firmware not supporting a mavlink message for these values. So you end up with -1 value shown (signals no value), as opposed to the value just not being available in the ui.

AndKe, 2016-12-04:
Please display current as amperes  (not mA)   - While monitoring a flight - one is usually just looking for unusual power, mA is harder to read/draws less attention when outside normal range.

DonLakeFlyer, 2016-12-04:
I"m trying to match the units for most batter capacity which is shown in mAh. Does that not make sense?

dogmaphobic, 2016-12-04:
I had the same issues early on when I first started. I think in Amps not Milliamps. At least not for circuits that operate in the 100A range :) I think the reason batteries show their capacity in mA is because big numbers look better :) 10,000 mAh looks a lot more impressive then 10Ah.

DonLakeFlyer, 2016-12-05:
I don't think in either amp or milliamps, so whatever you guys say is better is fine.

dogmaphobic, 2016-12-05:
Oh, I have no idea what the "standard" is. I just got used to it.

AndKe, 2016-12-05:
yes, consumed capcity is usually displayed as mAh - which is kind of good (unless the community switch to more correct Wh , which would then be more linear regardless of voltage)
Displaying mAh is kind of good, as observing the time 10 /100 mAh changes, gives an impression of discharge speed.

Seeing current, on the other hand - as in OSD or other GCSs (mavproyx, AP2, Tower, MP etc..)   is done in A. 
When on OSD, one may say it's to reduce display area, but most fly in the range on 10..60A   and noone is really paying attention to how many milliamps it is, as the throttle fluctuate more than hundreds mA very often.

DonLakeFlyer, 2016-12-05:
Got it. Will change.

DonLakeFlyer, 2016-12-07:
@AndKe Can you give this a try say tomorrow to make sure it is working correctly?

AndKe, 2016-12-07:
Yes, it works fine-.  On second though, I argued for making it like other GCS, - most seem to display two decimals. 
Soffy for the forth-and back, but if I make a PR for amperes with two decimals, will you pull it ? (or you can just do it)

DonLakeFlyer, 2016-12-07:
It's a one line change to BatteryFact.json. In the middle of something else right now. Can you do it?

AndKe, 2016-12-07:
Sure, I saw it near the changes you've made.

AndKe, 2016-12-08:
I changed types, learned from "battery" - still no success, current decimals are always "00". 
this is my (failed) attempt.  https://github.com/AndKe/qgroundcontrol/commit/041e3b9bd1cfa04f19036d068844f0f1d61bf1ce

I guess you see the reason right away :)

DonLakeFlyer, 2016-12-09:
Bummer. The code inside is converting things to int. I'll fix it.

closed 2016-12-22 by DonLakeFlyer