mavlink/qgroundcontrol #4232: Commands to vehicle sometimes appear to be lost

tfoote, 2016-11-22:
I'm seeing behaviors where I have to send commands to vehicles multiple times. The commands I'm testing with most are, "arm", changing flight mode, and "takeoff".

I'm running multiple vehicles in SITL. And it appears that the problem gets worse as the number of vehicles and load gets higher. Sometime it can take clicking in the UI 10+ times to send a command before it's received. Clicking and sliding arm, that many times is quite frustrating. 

I'm watching the mavproxy.py console and can see when commands fail, such as arming when it's not ready. But this problem is not that the command is being refused, but that the command is not even getting there to be refused. 

It sometimes seems like qgroundcontrol might be in the middle of doing other operations such as communicating the mission waypoints. 

Here's a video of the behavior with the consoles of all the mavproxy on the left and about 5 vehicles.
https://youtu.be/xP6CEFyIRsU

Once a command is received by mavproxy you can see it print in the console. If the command comes in and is invalid, mavproxy will display the error in the console. Unfortunately I didn't record an example of that. 

I may have some time to look into this but could use some suggestions for debugging. Is there a way to turn on a more verbose output related to communications?

As mentioned this is happening in SITL and on loopback or ethernet devices so I believe that the bandwidth should not be an issue.

DonLakeFlyer, 2016-11-22:
First thing can you make sure you can reproduce it with a daily build? The Stable build is getting a bit old now. Also does it reproduce without mavproxy connected?

DonLakeFlyer, 2016-11-22:
Wait is that the mavproxy that comes up with ArduPilot SITL?

DonLakeFlyer, 2016-11-22:
@tfoote I assume you are using ArduPilot SITL. Can you explain how you are running multilpe ArduPilot SITL instances so I can repro?

DonLakeFlyer, 2016-11-22:
Also just FYI: The codepath to get these messages out is pretty straightforward. It they really don't make it out the pipe then it is most likely a UDPLink problem. We have had problems with UDP links losing data in the past. Also currently sent commands are not resent if an ack doesn't come back for them. That needs to be changed as well. But that said over a SITL udp link packets should not be lost.

tfoote, 2016-11-23:
Yes, it's using ardupilot's SITL mavproxy setup. I just got pointed to the possibility of removing mavproxy from the loop: https://github.com/osrf/uctf/issues/50

There's instructions to launch multiple SITL's in this repo: https://github.com/osrf/uctf

I'm actively developing on the payload_v2 branch. 

If you have a xenial machine we have debs that you can dowload and run quickly following these instructions: https://github.com/osrf/uctf/tree/payload_v2/doc/install_binary  

This branch is under active development and is using the ardupilot by default and there's quite a bit of extra stuff being brought up on top that's only partially functional, but shouldn't interfere with the basic ground control interface to the drone. 

After that you can just 
With the environment setup you can
`roslaunch uctf uctf.launch gui:=false` and then in another window with the environment setup
`spawn_blue 1 2 3 4` to get 4 arducopters. Adding '--px4` will switch it to rotors SITL instances instead. I haven't tested the px4 version recently. 

And if you use blue, all the SITL instances can be connected to by the ground control on port 14000

DonLakeFlyer, 2016-11-23:
Wow, that's a pretty involved toolchain! I'd guess lots of possible failure points unrelated to QGC doing something wrong. I'm going to try to repro this with a direct path to multiple SITL instances.

DonLakeFlyer, 2016-11-23:
I can repro this just using a direct connection the 5 ArduPilot instances

tfoote, 2016-11-23:
great, let me know if I can help or test anything

DonLakeFlyer, 2016-11-23:
Likely same problem as #4235 MAVProxy issue.

DonLakeFlyer, 2016-11-23:
This is a problem with MAVProxy not sending commands to the right vehicle so they get lost. I've been able to get commands to go through to 3 different vehicels without problem by loading SITL directly without MAVProxy. For some reason anything beyond 3 fails to connect, but that also seems to be some sort of SITL vehicle problem as well. So nothing really going wrong here on the QGC side of things.

I'm going to keep this open since I seem to have lost the Issue about checking for command ack and retry. This will make the above problem a little better. But it will still happen.

DonLakeFlyer, 2016-11-24:
FYI: The problem with the 4th and 5th vehicle were setup issues. I can run 5 vehicles without problem.

DonLakeFlyer, 2016-11-24:
Command ack issue is #2975. Will fix in 3.2

closed 2016-11-24 by DonLakeFlyer