mavlink/qgroundcontrol #4253: Connection issue when QGC is started before USB connection

AndKe, 2016-11-26:
Please see the contradicting information about connection status: (it is not connected)

To reproduce: 
-Start QGC
-connect Pixhawk to USB
result: [image]

Now: 
-connect Pixhawk to USB 
-start QGC 
= success

-disconnect Pixhawk
-reconnect Pixhawk to the still running QGC
= same failure.

(tested on Ubuntu, using ArduPlane)

DonLakeFlyer, 2016-11-27:
I think you're running the version of ArduPlane that reports two different vehicle ids. At boot it sends on id, then eventually it sends the correct id. The connection list is from the bogus vehicle id, then the Vehicle 1 is from the real vehicle id which is eventually sent.

DonLakeFlyer, 2016-11-27:
--logging:MultiVehicleManagerLog will tell you

AndKe, 2016-11-27:
You are probably right, I used APM ArduPlane 3.7.0 for testing, last thing I did yesterday was to upgrade to AP 3.7.1 using QGC (which worked fine)

Anyway: the state QGC was in may need some attention, ("disconnect" button was not working, apparently no way to avoid a application restart)
- I am not aware of any way to choose between different systems (multiple vehicles/systems)  - that's not supported, right ?

DonLakeFlyer, 2016-11-27:
Due to the ArduPilot bug you are now running into: https://github.com/mavlink/qgroundcontrol/pull/4259. Looks like I'm going to need to fix that sooner rather than later.

> I am not aware of any way to choose between different systems (multiple vehicles/systems) - that's not supported, right ?

Yes it is. If you have multiple vehicles you get a drop down to select between them. It's just that #4259 is preventing you from getting at it.

DonLakeFlyer, 2016-11-27:
I believe tridge fixed the problem in 3.7.1

closed 2016-11-27 by DonLakeFlyer