mavlink/qgroundcontrol #4254: Some requirements for professional use: - (please consider adding)

AndKe, 2016-11-26:
For professional use, some features are needed, , I'll list them , and explain why:

- "preflight reboot", and/or "preflight calibration"  actions

[image]

The RPAS may be booted and prepared with payload tested, and awaiting ATC clearance or simply testing/calibrating payload for so long, that onboard sensors like barometer or IAS are not longer zeroed.
Another reason, while waiting, being on groundpower, it throws off the spent battery capacity (mAh - reported, and logged by the AP)   It's better to reboot the AP right before flight, thus resetting the total mAh statistics, then disconnect ground power.   

- "Trigger camera" action.
- useful for testing payload, and especially with TRIG/CAM , testing the whole loop for proper function. - this also allows us to verify that there is no dirt on the lens or protruding thing on the area where camera "sees out"  - before takeoff.  - again - for consistent logs, it's good to do a preflight reboot after taking pictures on ground.

-"Restart mission" ,and "go to waypoint"  mission controls
Very useful if a mission is to be paused , and holding at lower altitude due to other traffic.
It's often just as good to restart mission if it's recently started before such event , thus allowing more distance for the plane to climb and get properly on track.

When deep into a mission - it's better to resume using "go to waypoint" , (selecting a proper waypoint to allow climb)

-"Suspend Shooting" 
save current setting, and set CAM_TRIGG_DIST = 0;
-"Resume Shooting" 
restore last CAM_TRIGG_DIST   (may also be triggered is currently suspended and back in AUTO or "if reached a WP (mission resumed by "go to waypoint") )

This is useful when mission is paused, to avoid hundreds of photos taken while flying in a circle or holding pattern. 

Also , a simple "suspend shooting" and goto waypoint is useful when flying oversize mission, - when mapping bigger areas, or areas with strong winds, it's better to draw all-in-one, fly the endurance,  then, based on distance and wind conditions back home, stop shooting,  return, and resume with next wp of the same mission.  Such mission may end up taking only two flights , but if it were to be preplanned fully, and conservatively as separate missions, it may have been three or more flights.

dagar, 2016-11-26:
You can reroute to a specific waypoint if you click on it in the fly view and confirm. Selecting the first waypoint would be the same as restarting.

In general I would want the autopilot firmware to better handle these situations where you like to reboot, but I also want reboot for different reasons. Any opinion where we could hide the option?

AndKe, 2016-11-26:
> "You can reroute to a specific waypoint if you click on it in the fly view and confirm"

I failed to discover that, pretty neat.
As for the other actions,  they could be inside "fly view", near the flightmode menu, maybe called "CMD" beside mode like "...22.1v   RTL    CMD"    
, clicking it would open a list (like flightmodes), but with the actions mentioned, of course, some actions would require confirmation /(slide to confirm) or be unavailable if flying /(like calibration/reboot)

Alternatively, for better space and more expansion possibilities, maybe it should be placed under "vehicle setup" , under "summary" - something called "commands"  ?

Personally I'd prefer the first solution, "CMD" near flight modes.

tubeme, 2016-12-07:
@dager I think @AndKe has his point. I understand that the new trends are wiswig regarding the interface, but for professional use in more complex procedures, there should be a more formal way of issuing real time changes in the mission. So having the way to say "goto waypoint .... "Enter"" is much more simple and more operable in a situations when you fly a 150+ points mission and you get request/order to make some change in the mission in real time. In mission more than 60-70 points I get lost very often while editing, not to think if I have to move around the map during flight.

So in general not a bad idea to have some sort of ShortCuts to do commands and control.

dagar, 2016-12-14:
Just curious, but what's the use case for needing to actively jump between different waypoints like this? I only use it occasionally during testing.

Why are you editing so many waypoints? Can we do a better job with complex mission items so that more is done automatically?

AndKe, 2016-12-14:
@dagar  Let's say the mission is to map an area for a customer, like a dam or the area where rain is collected.  remaining daylight may be limited after travelling that day.  When the mission is planned, carefully altitude-adjusted (the terrain in northern Norway is not a exactly flat) - one need to draw it so the planes endurance  (~100minutes  or 80km flightpath) is taken into account. Crosstracking wind reduces the max flightpath.  Stronger wind also consumes more power as the plane tries to keep a good GS.

So:
I could plan the mission in parts, first far away from me, it must be calculated to do the grid in strong crosswind, then headwind for return.  this gives a conservative plan, when finished ,return, landing, battery change, and next part.  - each mission part needs to be verified with the previous for proper overlap.

It's much more efficient to plan it all as one mission:
Start far away, fly grid until remaining endurance demands return (depending on wind conditions), then return, and next flight resume from WP xx 
Resumed WP is set to last_completed_WP minus 4 - to create plenty of overlap data if the lightning conditions, shadows moved.

Splitting a mission conservatively for worst conditions may require (in a given scenario) four flights, if there's good weather that day, it may be possible to fly the same area in two flights.  - this is why one big mission is better that x small ones.

dagar, 2017-01-26:
@AndKe reboot - https://github.com/mavlink/qgroundcontrol/pull/4472

DonLakeFlyer, 2017-04-13:
> "Trigger camera" action.

Available in the instrument panel pages.

> "Restart mission" ,and "go to waypoint" mission controls

The "Start Mission" action will now pop up automatically if you pause a mission and will continue one to the next waypoint. Also if you RTL from the middle of a mission to say change battery. If you don't disconnect the connection in QGC when you pull the battery from the vehicle. When the vehicle comes back up Fly view will show you a Resume Mission option which will regenerate the mission from the place you left off and then you can take off and fly the remaining part.

flochir, 2018-11-08:
Got another detail problem on this. I flew a big multirotor survey and did a RTL to change the battery. QGroundcontrol offered me a mission resume, however I decided to take off manually due to the circumstances. After the manual takeoff, the resume offer was gone and I only got the regular "continue mission" which was set to the first waypoint again. That time, I didn't note down the last waypoint, so that was inconvenient.

It would be great to still be able to resume the previous mission after takeoff, ideally having the choice (Start over/resume).

closed 2026-10-04 by github-actions[bot]