mavlink/qgroundcontrol #4268: Replay Flight Data does not work well

jilinggui, 2016-12-01:
The log records saved by the qgc can not display the waypoint during playback, and some parameters are not fully loaded,someone can help me?Thanks!

DonLakeFlyer, 2016-12-01:
At this point log playback only shows vehicle telemetry. Missions and parameters will be flaky.

DonLakeFlyer, 2017-05-17:
We'll clean all this up on 3.3

Antiheavy, 2017-10-05:
Missions and parameters would be super useful to have in the ground station log if that is technically feasible.

DonLakeFlyer, 2017-10-06:
Saying the "Replay does not work well" is quite an understatement! It's pretty horrendous.

DonLakeFlyer, 2017-10-06:
> Missions and parameters would be super useful to have in the ground station log if that is technically feasible.

Yes it is. And I've been meaning to do it for ages. Non-trivial coding-wise but great feature.

DonLakeFlyer, 2018-06-20:
I tried support parameter and mission pulling from logs but it turned into a nightmare. For now Replay is working with recent changes.

closed 2018-06-20 by DonLakeFlyer