mavlink/qgroundcontrol #4269: Probelms with Pitch, Roll, Yaw, and Throttle coupling in Joystick Calibration

dbarge, 2016-12-01:
Hello,

There is an inherent problem in the joystick calibration procedure in QGC.
Because the coupling procedure is based on user input from the joysticks (in my case the Logitech F310), there will always be some amount of coupling between Roll. Pitch, Yaw, Throttle, etc.
For example, when I calibrate roll I can see that the pitch controls move slightly in the "Axis Monitor".
The same is true for other controls coupled via the joystick inputs.

The QGC joystick calibration tab is really user friendly and you can get good results with it really quickly.

Unfortunately these couplings can cause problems either when trying to fly for the first time with a standard airframe or when trying to develop custom airframes.
They can also add small correlations (between roll and pitch for example) even in working airframes that result in (small) degradations to performance.

What would be great is if there was a way to tighten these up (by hand if needed) in the QGC UI.
From the current calibration tab, QGC can determine the directions and limits of the joystick with certainty.
For example, if I pitch forward with the right joystick (on a Logitech F310) even if there is a little roll (because I'm not coordinated enough to push the right joystick perfectly in the pitch forward direction) it should still be easy to tell the true pitch direction because the magnitude of the joystick signal in the pitch direction is much greater than the corresponding roll signal.
So the direction can be determined just by taking the max of the two signals and that should be enough to remove the directional coupling.

It should also be easy to find the maximum magnitude of a particular joystick signal.
For pitch for example, if the user pushes the joystick to where they think full pitch is but there is a slight roll component, the pitch magnitude will be a little less than it's max value.
But if the user pushes the pitch control to where they think full pitch is and then they roll it around (at the limit of the joystick articulation) slightly it should be easy to find the maximum pitch magnitude.

Is it possible to use such a procedure to fully decouple the joystick controls?

Thanks!

DonLakeFlyer, 2016-12-01:
@rjehangir Isn't there deadband on joysticks now?

dbarge, 2016-12-01:
I think so but this would still be an issue outside of the deadband.

DonLakeFlyer, 2016-12-01:
Are you talking about calibrating or while flying? If the calibration is slightly off from a min/max value standpoint in a stick orientation how does that affect flight?

dbarge, 2016-12-01:
Are you referring to deadband in the joystick center?

Calibrating...
The issue I'm talking about was still present in Stable_V3.0.
I think you can see the coupling in the calibration procedure.
It has a small effect on flight and can have an effect when testing custom mixers.
For example if you calibrate the throttle with a yaw component and then push the left joystick straight up in manual mode you will see the coupling. Of course it can be compensated for.
I wouldn't call it a high priority but it would be nice to have the controls fully decoupled.

DonLakeFlyer, 2016-12-01:
I guess I'm not getting it. Calibration does nothing to couple the joystick axes in any way. It only records the max/min for each stick direction. For example when it records the maximum value for the up position of a stick it records nothing about left/right position. If you push your joystick straight up and you see some left/right movement that has nothing to do with QGC. That is just the joystick not being very precise. Flying with a joystick is always going to not be precise, the sticks are just too small.

dbarge, 2016-12-01:
"For example when it records the maximum value for the up position of a stick it records nothing about left/right position."

This is the part I wasn't sure about... 
I was working with a custom mixer and trying to figure out what part of the imprecision was innate to the joystick and what part of the imprecision was due to the calibration procedure. You're telling me there's no coupling in the calibration procedure.

Closing... thanks.

closed 2016-12-01 by dbarge

DonLakeFlyer, 2016-12-01:
> You're telling me there's no coupling in the calibration procedure.

Yup. Correct.

pritamghanghas, 2016-12-01:
As Don said there is no coupling from QGC side. The real issue is these joysticks are not meant for rc control. I face this problem myself. I like the gamepad form factor but these sticks have very little travel on their stick potentiometers before they reach max. Making them very sensitive. Bad news is "there is no gamepad that has RC like sticks that allow for smooth longer travel. No one built them." Unless you use one of the full size USB joystick. I think they call them "USB RC simulator radio". They have RC like sticks but are big and new Bluetooth option.

dagar, 2016-12-01:
You can also connect a Taranis to a computer to use as a joystick.