mavlink/qgroundcontrol #4275: Hotkey to add current Pos as Waypoint

tops4u, 2016-12-01:
While flying survey Missions it would be good to have a Hotkey (on Computers) or a Softkey (on Mobile Devices without Keyboard) in order to mark the UAVs current Postion as a future Waypoint. So that the mission (Survey) can be continued while interesting areas can be marked for a visit after the survey mission. 

This type of Waypoint should be possible to preconfigure in advance (i.e. height, etc) .

tops4u, 2017-02-02:
@DonLakeFlyer : It would be very nice to have this feature within the next 3-4 Weeks. This could also be a crude 1st Implementation that just adds a Waypoint out of the current position without any Preconfiguration.

tops4u, 2017-05-18:
@DonLakeFlyer Is this no longer planed for any upcomming Release?

DonLakeFlyer, 2017-05-18:
Not in 3.2

tops4u, 2017-05-18:
Oh, pitty. This would have been very handy right now to search for fawn.

closed 2026-10-05 by github-actions[bot]